PiSupply/PaPiRus #126: Running the 2.7" Screen on the pi zero phat

uselesshippy, 2017-06-13:
Is it possible to run the 2.7" screen on the phat for the pi zero rather than the full pi version.

I want to do this as the full size hat uses pins I want to use for other phats, but the pi zero phat doesn't use them, specifically BMC18.

Thanks

uselesshippy, 2017-06-13:
Alternately can I change the underlying code for the Papirus to use a different pin and then wire that pin to the correct pin on the board?

shawaj, 2017-06-13:
@uselesshippy yes it is entirely possible to use any of the screens with any of the boards. Just insert the 2.7" screen as you would any other, and then in the software just run `sudo papirus-set 2.7` and it should just work with the PaPiRus Zero pHAT.

@francesco-vannini do you have any idea why PaPiRus HAT uses BCM18 and PaPiRus Zero doesn't? Is this definitely correct?

Thanks!

shawaj, 2017-06-13:
@uselesshippy @francesco-vannini just checked it - for some reason the HAT has a PWM output to the display. 

I have a funny feeling that this is not needed on the V231 film, as it is not on the Zero board, so perhaps it is something left over from our legacy support of the older displays.

@repaper @CharmingSu do you know about this at all? Would it cause malfunction if @uselesshippy just cut the trace from this pin and did not use it?

uselesshippy, 2017-06-13:
That would be fantastic if I could just do that.

repaper, 2017-06-13:
Please visit [this link](http://www.pervasivedisplays.com/products/271#aurora-mb-v231) to download the COG driver interface. For V231 with G2 driver IC, the PWM is no longer needed in the driving circuit. The Appendix at the last two pages has the explanation.

shawaj, 2017-06-13:
@repaper can you look at the schematic [here](https://github.com/PiSupply/PaPiRus/blob/master/hardware/PaPiRus%20HAT/Latest%20Version%20-%20v1.9/2014-035-01-Pi-ePaper-circuit_v1_9.pdf) and let me know if we can just not connect the PWM pin and for it to still work?

@uselesshippy which was it that you wanted to do? Use the PaPiRus Zero or cut a trace on the HAT?

uselesshippy, 2017-06-13:
@shawaj If I can cut the trace on the HAT without affecting functionality that would be my preferred solution. I was only going down the PaPiRus Zero route as I know that the pHat doesn't use that pin.

shawaj, 2017-06-13:
Hopefully @repaper can take a look and clarify :-)

We will also test in the office this week if we get a chance

shawaj, 2017-06-14:
@uselesshippy - our main man @francesco-vannini just tested this and it works perfectly without BCM18 with the 2.7" screen. We did it by getting a 40 pin through hole extension header and "cutting" pin 12 out.

closed 2017-06-14 by shawaj

uselesshippy, 2017-06-14:
Fantastic thank you for all of your help.

shawaj, 2017-06-14:
@uselesshippy - no problem!

Have added it to the hardware readme here too - https://github.com/PiSupply/PaPiRus/blob/master/hardware/PaPiRus%20HAT/README.md#pwm-pin---bcm-18--pin-12