PiSupply/PaPiRus #129: Removing multiple positioned text items and writing all new ones

eat-sleep-code, 2017-06-16:
I have a program that alternates between three screens.

Each of these screens has multiple items that are positioned at very places on the screen.

How do I smoothly clear the screen, and then write the next one.

I tried `display.Clear()` and then (after a bunch of `display.AddText`s) a `display.WriteAll()` but this causes the screen to blink twice between each screen.  

I tried a bunch of `display.RemoveText('ID')` and then (after a bunch of `display.AddText`s) a `display.WriteAll()` but this leaves large chunks of unwritten screen.

So, what is the cleanest and quickest way to accomplish a Write of all new positioned text.

shawaj, 2017-06-16:
The only way to properly clear the screen is with it blinking twice as you say.

You could try and write the inverse colour of the original to the same space but I think that might cause some ghosting and wouldn't give a truly clear screen.

eat-sleep-code, 2017-06-16:
Do you think the library could be improved, to possibly allow it to clear and replace the screen contents in one action?

I am just thinking on my Kindle, there isn't a big double flash like that when I switch pages.

shawaj, 2017-06-16:
It's a hardware limitation not a software limitation so I don't think there is anything we can do.

Unless @repaper or @tvoverbeek or @francesco-vannini have any ideas?

eat-sleep-code, 2017-06-24:
I notice that this code in textpos.py seems to write a new image and then clear the display?  

```def Clear(self):
        # Clear the image, clear the text items, do a full update to the screen
        self.image = Image.new('1', self.papirus.size, WHITE)
        self.allText = dict()
        self.papirus.clear()
```

Shouldn't just one of these be necessary?

shawaj, 2017-06-24:
@eat-sleep-code are you seeing a specific error with part of this code?

eat-sleep-code, 2017-06-27:
No error, just trying to track down why there are multiple flashes between each update.  If I look at how my Kindle Paperwhite (understand that the hardware is different, but both are e-ink so should operate roughly the same) there is a single flash between updates, not multiple.

shawaj, 2017-06-27:
What coffee are you running that is causing the behaviour you are seeing?

Need to be able to replicate it.

shawaj, 2017-06-27:
*code

eat-sleep-code, 2017-06-27:
Well it is a dark roast.   ;-)  

I will post the code into a repo tonight.

shawaj, 2017-06-27:
Thanks very much that'll be great

eat-sleep-code, 2017-06-28:
Here you go: https://github.com/eat-sleep-code/baseball-pi

Note: To see this in action, find a team that currently has a game in progress: https://www.mlb.com/

shawaj, 2017-06-29:
@eat-sleep-code and which line in there is the issue?

eat-sleep-code, 2017-06-29:
In [baseball-pi/baseball/functions.py](https://github.com/eat-sleep-code/baseball-pi/blob/master/baseball/baseball.py)  -- which I just cleaned up a bit...

Look at the `displayBoxScore` (line 135), `displayPitch` (line 161), or  `displayPlay` (line 176).

You will see I perform a `display.Clear()` near the beginning and a display.WriteAll() at the end of each of these.   

When a game for the selected team is in session, `displayInGame` (line 210) cycles through these methods every "5 seconds".  

In between each screen being displayed the screen flashes multiple times before the content is written.

tvoverbeek, 2017-07-01:
To achieve what you want, use partial_update.
Instead of display.Clear() only create an empty image and zero the
dictionary.(only lines 170-171 of Clear() in textpos.py)
When calling WriteAll use WriteAll(partial_update=True).

Look at the code for the clock or the game of life demo for a demo of
partial update,
Hope this helps

Ton van Overbeek
(Currently no access to RPi/PaPiRus since holidaying down under)

eat-sleep-code, 2017-07-02:
When I tried this I get the following error:

`TypeError: WriteAll() got an unexpected keyword argument 'partial_update'`

tvoverbeek, 2017-07-02:
Try WriteAll(True) instead. Look at the definition of WriteAll in
papirus/textpos.py.

eat-sleep-code, 2017-07-02:
Actually, I discovered I had a slightly older version of the library.   `partial_update=True` works with the most recent version.   Now, if issue #130 was fixed, my app would be working pretty much perfectly.

closed 2017-07-02 by eat-sleep-code