PiSupply/PaPiRus #130: Text incorrectly wraps with custom font...

eat-sleep-code, 2017-06-16:
- `display.AddText(text = 'Awaiting player roster', x = 5, y = 5, size = 14, Id = 'Status')` results in text wrapping because there is not quite enough room to fit the text with the default font.

- `display.AddText(text = 'Awaiting player roster', x = 5, y = 5, size = 14, Id = 'Status', font_path = '/font/benton-sans-regular.ttf')`  it still wraps, even though the font face should provide plenty of room for the text.

shawaj, 2017-06-16:
Thanks, this is a good find! As you say it's not actually taking into account the font geometry. 

@tvoverbeek do you know if there is a way to allow it to be aware of the size of the font?

eat-sleep-code, 2017-07-02:
This is pretty much the last thing standing in the way of my app working perfectly.

tvoverbeek, 2017-07-03:
Look at the PIL ImageFont documentation.
I believe the function you need is ImageFont.ImageFont.getsize(text) which
returns (width, height).

eat-sleep-code, 2017-07-03:
@tvoverbeak So that would get me height and width of my text?  But what would I do with the height and width?

tvoverbeek, 2017-07-03:
Look at the code in textpos.py (https://github.com/PiSupply/PaPiRus/blob/master/papirus/textpos.py).
Function addToImageText (lines 77-157) does the calculations for wrapping the text.
Its calculations are based on a fixed width font (equal width for each character).
This function needs to be modified substantially to support variable width fonts by using ImageFont.ImageFont.getsize(text). You will either have to do this yourself, or wait till I am back home in Europe after my holidays (after July 11) before I can experiment and test this.
Maybe somebody else beats me to it.
Regards from down under.

eat-sleep-code, 2017-07-03:
I will let you tackle it when you get back.  My Python skills are elementary, so would prefer if a pro did it.  Have a good holiday.

eat-sleep-code, 2017-07-20:
@tvoverbeek Had any chance to review this yet?

tvoverbeek, 2017-07-20:
@eat-sleep-code Back home now. Plan to start on it next week.

tvoverbeek, 2017-07-24:
See my PR #142. Should solve your problem.

eat-sleep-code, 2017-07-25:
@tvoverbeek see my comment on the PR about an extraneous preceding space.

eat-sleep-code, 2017-07-25:
Also might be nice to have option to limit text to _X_ lines (and truncate any remain data).

tvoverbeek, 2017-07-25:
See reply on the PR.
Regarding limiting the text, the code already limits the text to what fits
on the screen.
Any text going (vertically) off screen is discarded.

eat-sleep-code, 2017-07-26:
See pull request for text limiting.   In the case of my MLB app, sometimes a play description provided by MLB was too long and was colliding with the ball and base display below it.   By adding a maxLines feature, I can limit it to 3 lines and prevent the collision.   I also resolved the preceding space issue that was causing subsequent lines to be slightly updated.

tvoverbeek, 2017-07-26:
PR #144 has been merged. Closing issue.

closed 2017-07-26 by tvoverbeek