PiSupply/PaPiRus #132: Inverted picture pixels

shawaj, 2017-06-19:
On a similar note to #80 it would be really cool to have an "inverse" function for papirus-draw that takes a picture, makes it black and white (as it does now) but then prints the inverse of it... So what would normally be black becomes white and vice versa

schwma, 2018-03-16:
Note: It is also possible to invert colors in one line by letting ImageMagick invert the image and piping its output into papirus-draw.

Here is an example:
`curl https://raw.githubusercontent.com/PiSupply/PaPiRus/master/bitmaps/papirus-logo.bmp | convert - -negate - | papirus-draw - -t resize`

However, when using `-t resize` the background surrounding the image's bounds is always rendered in white (which may or may not be desirable).

closed 2018-03-17 by tvoverbeek

shawaj, 2018-03-17:
@tvoverbeek should we add the note above from @schwma to the readme do you think?

tvoverbeek, 2018-03-18:
@shawaj There is always more than one way to do things, especially in Unix/Linux.
To do the inversion on the black & white image is more efficient than to do it on the (possibly) full color image as is the case when using ImageMagick convert.
Hence, at least for me, no need to add this to the README.

commit 6bc861ed (2018-03-17): Add invert functionality to papirus-draw (closes #132) (#181)

* Add invert functionality to papirus-draw (closes #132)

* Update papirus-draw usage documentation