PiSupply/PaPiRus #136: Kernel Panic (Pi Zero W)

fabiocody, 2017-07-16:
I just installed the suite using the auto-installation, but now my Raspberry Pi Zero W is not booting anymore: after it starts the NTP service, it begins fast printing a bunch of numbers (I think it's a memory dump) and IF it ends printing, it says "kernel panic - not syncing: fatal exception in interrupt". Before doing the auto-installation, I only upgraded the system (which was fresh clean) and turned on I2C and SPI.

francesco-vannini, 2017-07-17:
Hi @fabiocody I am really not sure this could be associated with the installation of PaPiRus' code. Have you tried installing on a different Pi? Have you soldered the header yourself on the Pi?

fabiocody, 2017-07-17:
This is the only Pi I have installed the code on and yes, I soldered the header by myself. But I think it's worth noting that the Pi works absolutely fine (and the PaPiRus HAT too) if I install the code manually, so I definitely think it's something related to PaPiRus auto-installation process.

francesco-vannini, 2017-07-17:
@fabiocody thanks for clarifying this. As soon as I have a spare second I'll try this myself. I'd be surprised if it turns out to be our code but you never know.
Would you be able to show me the output of your installation somehow? If you are remoting into the Pi you should still be able to copy/paste things even after the connection is dropped.

francesco-vannini, 2017-07-17:
@fabiocody I have just tested on a fresh install and all is working fine for me. By all means get back on this thread but I'll go ahead and close it because it's not an actual problem with the code.

closed 2017-07-17 by francesco-vannini