PiSupply/PaPiRus #139: Any guide for production use?

CraigglesO, 2017-07-21:
I was hoping on making a device (for fun) using only the screen. Is there any resources you could provide for this?

shawaj, 2017-07-21:
You can't use only the screen, you would need some kind of microcontroller
to update what's on it.

What's the use case you had in mind?

CraigglesO, 2017-07-22:
Ah, ok. I see. I was hoping to keep the device as slim as possible. have the screen practically flush with the raspberry pi zero. It's seriously the perfect size, and I just find the e-paper so beautiful. But, with the headers and all, the thickness is a little too much for my taste.

If there was a design that had the pins to connect directly from the back side, that would be enough of a low profile I think. I just don't have enough knowledge of the project.

shawaj, 2017-07-22:
Have you tried the Pi Zero one? Or perhaps the HAT version with our
official case for it?

CraigglesO, 2017-07-23:
[image]

This is my setup.

I'm being too picky. This is great. I just got a 3d printer, and I'll make a case with this setup :)

shawaj, 2017-07-23:
You could try removing the plastic from the header and cutting the pins
shorter?

CraigglesO, 2017-07-23:
Huh, yeah. I should try that! Honestly, that should work fine! I'll close this as it is solved.

closed 2017-07-23 by CraigglesO