PiSupply/PaPiRus #140: use PapiRus on tinkerboard.

alr2000, 2017-07-21:
Hi!
I would like to use the Papirus screen on a Tinker Board. Could be possible?
I´m pretty new at linux and python, so could you do a few hints to work on?

Thanks!

tvoverbeek, 2017-07-21:
Should be possible. Tinker Board TinkerOS is also based on Debian ARM as is Raspbian.
I did a quick compare of the Tinker Board 40-pin GPIO connector and the RPi 40-pin GPIO connector. For the Papirus HAT you need I2C and SPIO. Both are on the same pins. They obviously tried to be as compatible as possible with the RPi.
Software wise you need to port the driver from the repaper repository (https://gitub.com/repaper/gratis). Probably works without changes.
Good luck (but you'll need a bit more knowledge about Linux than a newbie)

alr2000, 2017-07-24:
Dear tvoverbeek,
Thanks for the swift reply. 
I´m taking a look to the repaper repository, I realized that I should create a new folder inside the gratis/PlatformWithOS folder with the Tinkerboard specific data, and also modify some make files to allow the compilation with the new platform. Could you point the important steps that I should follow?

Thanks.

P.S. Maybe I should post this topic in the repaper repository to get help from them.

tvoverbeek, 2017-07-24:
@air2000, Suggest to copy the RaspberryPi directory to a new Tinkerboard directory and check
the gpio and spi definitions and modify for tinkerboard. Probably not much is needed since the pin layout is almost identical to the RaspberryPi one. For spi you will have to llok at the TinkerboardOS specifics.
HTH.

shawaj, 2017-07-24:
I've ordered a couple tinker boards to test with but doubt I'll get around
to anything anytime soon. If I do I'll post back here.

alr2000, 2017-07-25:
I did some improvements. The screen is reacting to the communication, but isn´t accurate.
I used a GPIO C library(https://tinkerboarding.co.uk/wiki/index.php?title=GPIO) instead of change all the raspberry registers. I change the content of the functions on PlatformWithOS/RaspberryPi/gpio.c to use the library. Also I change the PlatformWithOS/RaspberryPi/epd_io.h to match the new ping numbering and my spi device on /dev/spidev2.0.

This is the result using the PlatformWithOS/driver_common/epd_test compiled with 
`make rpi EPD_IO=epd_io.h PANEL_VERSION='V231_G2'`
[image]

Any hints?

alr2000, 2017-07-25:
I check the pins with a oscilloscope and I get these results:
The first image correspond with the clean and image0 of the epd_test 
[image]
The second image correspond to a detailed clean.
[image]
YELLOW= Discharge
BLUE= Panel
GREEN = Reset
PINK= Border.

I´m going to try with a raspberry to see if the signals match in the two devices.

shawaj, 2017-07-25:
Have you set the screen size to 2.0 inch in the driver?

alr2000, 2017-07-25:
@shawaj No, I used only `PlatformWithOS/driver_common/epd_test 2.0` to test it.
Where should I change?

shawaj, 2017-07-25:
That should have worked if you did EPD test with 2.0 after.

It's getting communication which is a good sign but it's not displaying the
correct information.

@repaper @charmingsu have you ever seen an issue like this before?
Wondering what the issue could be?

alr2000, 2017-07-25:
Seem to be that GPIO signals are fine:
[image]
[image]
I think should be the SPI.

alr2000, 2017-07-25:
I think I found the problem, but I couldn´t solve yet.
The problem is related with GPIO library.
The gratis code uses 5 pins(4 output(border,reset,panel and discharge) and 1 input(busy)). I realized that the busy pin is not configured as INPUT when I run the edp_test, is configured as OUTPUT as the all others pins.
But the problems is that when I try to modify the pin mode via the `echo input >/sys/class/gpio/gpio171/direction/` the all others pins also change the mode and viceversa.
[image]
[image]

`

alr2000, 2017-07-26:
Hello again,
Forget the last message. I did a deeper research into the status of the tinkerboard gpio register and they are right, but the GPIO library doesn´t show the right status of the pin.

I did a double check of pin´s behavior in both devices and I discovered this difference in the EPD_CS_COG pins:

Yellow: TinkerBoard
Pink: Raspberry pi.

[image]

alr2000, 2017-08-02:
I found another diference between the raspberry and the TinkerBoard SPI bus.
I send 3 bytes[0x80,0x82,0x81] with mode 0 and bit 8 bits per word, with no delay.
This is the raspberry spi .
[image]
This one is the tinkeboard 
[image]

As you can seen the raspberry does a clock pause between each byte, but the tinkerboard doesn´t do this pause. Could it be the problem? I don´t now how sensitive is the clock pin on the display, but according with the DS, the behavior should be like the raspberry.
[image]

shawaj, 2017-08-02:
Could you open an issue at https://github.com/repaper/gratis referencing
this issue just so it gets the attention of someone from Pervasive?

Also copying in @repaper @charmingsu

alr2000, 2017-08-02:
Thanks, I already did it. I hope they can help.

shawaj, 2017-08-02:
The screen is obviously getting some data as it's changing the screen which
is promising but annoying!

Hopefully it's a fairly simple fix :-)

shawaj, 2017-08-02:
@tvoverbeek do you have any ideas?

Do you have a Tinker board?

tvoverbeek, 2017-08-02:
Aaron,

No, I do not have a Tinkerboard.
Do not know if they have any support forums which might help with the
Tinker board SPI details.

Ton

shawaj, 2017-08-02:
> issuecomment-319652851

@tvoverbeek shall I send you a Tinker?

@alr2000 do you know if there's a forum or support area for Tinker?

alr2000, 2017-08-03:
Dear @shawaj , yes there is an area in https://tinkerboarding.co.uk/forum/forum-1.html.

alr2000, 2017-08-16:
Ok, I found the problem.
It was the SPI of the tinkerboard. If it uses the DMA it sends random data beyond 32 bytes.(each scren row is 76 bytes long for the 2.0 inch). So you have to disable the spi dma in the kernel. More info on:
https://github.com/rockchip-linux/kernel/issues/19

shawaj, 2017-08-17:
> rockchip-linux/kernel#19

Thanks so much for your investigation work here!

So you have a guide on how to build this in full?

alr2000, 2017-08-18:
1° Get the kernel sources from https://github.com/TinkerBoard/debian_kernel
2° Modify /drivers/spi/spi_rockchip.c according to rockchip-linux/kernel#19
3° Compile the kernel:
`sudo make miniarm-rk3288_defconfig`
`sudo make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf-`
4° Substitute the new kernel(zImage) in the /boot folder of the current TinkerOS system.
5° Reboot
6° Get gratis Library
`git clone https://github.com/repaper/gratis.git`
7° Download and install GPIO C Library for Tinkerboard according to  https://tinkerboarding.co.uk/wiki/index.php?title=GPIO
8° Create a new folder for Tinkerboard system on PlatformWithOS
9° Modify the Makefile to allow the tinkerboard code and include the wiringPi library on it.
9° Add to the Tinkerboard theses files
`epd-fuse.sh  epd_io.h  epd_io_free_uart.h  gpio.c  gpio.h`

**epd_io.h**

```
#if !defined(EPD_IO_H)
#define EPD_IO_H 1

#define panel_on_pin  4
#define border_pin    15
#define discharge_pin 16
#define pwm_pin       2
#define reset_pin     5
#define busy_pin      6
*/
#define SPI_DEVICE    "/dev/spidev2.0"
#define SPI_BPS      8000000

#endif
```
**gpio.c**
```
// Copyright 2013-2015 Pervasive Displays, Inc.
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at:
//
//   http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing,
// software distributed under the License is distributed on an
// "AS IS" BASIS, WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either
// express or implied.  See the License for the specific language
// governing permissions and limitations under the License.

// For the BCM SOC Preipheral Manual register layout see:
//   http://www.raspberrypi.org/wp-content/uploads/2012/02/BCM2835-ARM-Peripherals.pdf
//
// also see: http://elinux.org/RPi_Low-level_peripherals
//
// Other items are difficult to determine accurately. Google search "Gert Doms" for some samples

#include <stdint.h>
#include <stdbool.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <ctype.h>
#include <fcntl.h>
#include <sys/mman.h>
#include <unistd.h>
#include <err.h>
#include "gpio.h"
#include <wiringPi.h>

bool GPIO_setup() {
        wiringPiSetup ();
        return true;
}

// remove the map
// revoke access to GPIO and PWM
bool GPIO_teardown() {
        return true;
}

void GPIO_mode(int pin, int mode) {
        pinMode(pin,mode);
}
int GPIO_read(int pin) {
        return digitalRead(pin);
        }

void GPIO_write(int pin,int value)
{
        digitalWrite (pin, value);
}
```
**gpio.h** 
```
// Copyright 2013-2015 Pervasive Displays, Inc.
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at:
//
//   http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing,
// software distributed under the License is distributed on an
// "AS IS" BASIS, WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either
// express or implied.  See the License for the specific language
// governing permissions and limitations under the License.

#if !defined(GPIO_H)
#define GPIO_H 1

#include <stdbool.h>
#include <stdint.h>

// GPIO modes
typedef enum {
        GPIO_INPUT,   // as input
        GPIO_OUTPUT,  // as output
        GPIO_PWM      // as PWM output (only for P1_12
} GPIO_mode_type;

// functions
// =========

// enable GPIO system (maps device registers)
// return false if failure
bool GPIO_setup();

// release mapped device registers
bool GPIO_teardown();

// set a mode for a given GPIO pin
void GPIO_mode(int pin, int  mode);

// return a value (0/1) for a given input pin
int GPIO_read(int pin);

// set or clear a given output pin
void GPIO_write(int pin, int value);

// set the PWM ration 0..1023 for hardware PWM pin (GPIO_P1_12)
void GPIO_pwm_write(int pin, uint32_t value);

#endif
```

The rest of the files stay the same.
10° Compile and test the epd_test.c
[11°] Get the PaPiRus code and start playing!

shawaj, 2017-08-21:
@tvoverbeek @francesco-vannini what do you think about this? 

Looks like mostly changes to RePaper GitHub but probably worth us formally doing this as the PaPiRus codebase itself will stay the same

shawaj, 2017-12-23:
Just FYI - the Asus Tinker staff are looking at this as we speak