PiSupply/PaPiRus #143: Inconsistency in variable format?

eat-sleep-code, 2017-07-25:
I note in the code, that there seems to be an inconsistency in variable format?

For example:

`def UpdateText(self, Id, newText, font_path='/usr/share/fonts/truetype/freefont/FreeMono.ttf')`

Here there appears to be 3+ different variable formats:

`self` and `newText` appear to be CamelCased.
`Id` appears to be PascalCased
`font_path` appears to be -- yeah, I don't know what you call that, underscoreCased? 

Is there a reason for this?

tvoverbeek, 2017-07-25:
No specific reason, except different persons with different styles
contributing to the code.
There is no PiSupply coding style document (yet?).

Regards

Ton
PS Is your MLB program working as you want now with this change?

eat-sleep-code, 2017-07-25:
Other than the preceding space issue (will look into that tonight).

shawaj, 2017-07-25:
Yeah we're not too picky here about that stuff. I guess we're not purists
like that. I'm not a software engineer though, I'm sure if I was it would
probably bug me a little.

Happy to decide on one if you wish @tvoverbeek but seems like extra work
for no real gain to me.

What do you think also @francesco-vannini ?

eat-sleep-code, 2017-07-25:
It is just the OCD in me.   From a code maintainability aspect, it is definitely easier to follow if it is consistent.   Plus, I wanted to know which one I _should_ use if I wanted to make any tweaks.

eat-sleep-code, 2017-07-26:
See pull request that fixes this and a couple other issues.

tvoverbeek, 2017-07-26:
Merged PR #144. Closing issue.

closed 2017-07-26 by tvoverbeek