PiSupply/PaPiRus #145: Unexplained text wrapping due to embedded space character

kennovation1, 2017-07-26:
I'm creating a composite screen with a 2.7" PaPiRus and when I add text that has an embedded space, the space appears to be treated as a newline. If I replace it with an underscore, it works fine.

Relevant Python 2.7 snippets...
```
disp = PapirusComposite(False)
disp.AddText("NO STATUS", 90, 10, Id="someID")
disp.WriteAll(True)
```
There is plenty of space on the right side of the text, and of course the width is roughly the same if I use an underscore instead of a space. I added a space in other text that appears on the display and that works fine. I tried invert=True and invert=False. I tried setting font_path to a couple of other fonts and that had no effect.

Another possible clue is that I then do an UpdateText with the string "RUNNING" (using the same ID). This renders fine, but the errant second line from where the STATUS string incorrectly appeared gets erased as expected, but leaves a white rectangle erasing the other text that was in that same location that the status line overlapped.

If I use a long string without spaces it works fine "RUNNING1234". If I shorten to "NO STAT" it works (no wrap), but "NO STATU" does wrap.

Seems like a bug, but please let me know if I'm just doing something brain dead.

Thanks!

kennovation1, 2017-07-27:
I found it! This is indeed a bug. From /usr/local/lib/python2.7/dist-packages/papirus/textpos.py line 117:
`elif (x + draw.textsize(text_lines[current_line] + " " + word, font=font)[0]) < line_width:`
should be
`elif (draw.textsize(text_lines[current_line] + " " + word, font=font)[0]) < line_width:`

The `x +` shouldn't be there since the x offset is already accounted for in line_width (which is screen width minus x).

Can you fix please?

tvoverbeek, 2017-07-27:
Good catch !!.
Thanks.
Fixed now in master.

closed 2017-07-27 by tvoverbeek