PiSupply/PaPiRus #146: All scripts end in error: No such file or directory

f4exb, 2017-07-28:
Example:

sudo papirus-clock 
Traceback (most recent call last):
  File "/usr/local/bin/papirus-clock", line 111, in <module>
    main(sys.argv[1:])
  File "/usr/local/bin/papirus-clock", line 52, in main
    papirus = Papirus(rotation = int(argv[0]) if len(sys.argv) > 1 else 0)
  File "/usr/local/lib/python2.7/dist-packages/papirus/epd.py", line 58, in __init__
    self._lm75b = LM75B()
  File "/usr/local/lib/python2.7/dist-packages/papirus/lm75b.py", line 21, in __init__
    self._bus = smbus.SMBus(busnum)
IOError: [Errno 2] No such file or directory

f4exb, 2017-07-28:
Well I looked at issue #124 (closed now) and eventually I got i2cdetect showing address 0x48. Then I got papirus-clock working.

I use a RPi2 with the 40 pin connector. I unplugged and re-plugged the Papirus maybe that did the trick.

closed 2017-07-28 by f4exb

shawaj, 2017-07-29:
Have you enabled i2c and SPI?

neuberfran, 2019-04-12:
[image]
anyone

shawaj, 2019-04-12:
@neuberfan - are you using a PaPiRus?

neuberfran, 2019-04-12:
yes, if I use 0x1b(7bits) works, but I need 0x36 (8bits)
[image]

shawaj, 2019-04-12:
What are you actually trying to do? And you are using a PaPiRus epaper display yes? 

Please start a new issue explaining what you are trying to do, what software you are using on your Pi, what version of the PaPiRus you have and what issues you are facing.

tvoverbeek, 2019-04-12:
@neuberfran 8 bits addressing does not really exist in I2C.
The most significant 7 bits are the address and the least significant bit indicates write(0) or read(1).
So your 8-bits 0x36 is in fact a write to 7 bits address 0x1b.
Moreover the PaPiRus has nothing on address 0x1b. It uses SPI to drive the display. The HAT RTC uses (7-bits) I2c address 0x6f.
So you must have something else on address 0x1b. Hence not a PaPiRus issue at all