PiSupply/PaPiRus #148: SyntaxError

Sharkeon, 2017-08-12:
When it try to run the code:

from papirus import PapirusText

text = PapirusText([rotation = rot])

text.write("hello world")

I get this Error: 
File "test.py", line 4
    text = PapirusText([rotation = rot])
                                                ^
SyntaxError: invalid syntax

shawaj, 2017-08-12:
Where it says rot you need to put a number from 0 the 360 for the angle

Sharkeon, 2017-08-12:
I tried. Still getting the same Error

shawaj, 2017-08-12:
Did you remove the square brackets too?

Sharkeon, 2017-08-12:
now i get this Error:

Traceback (most recent call last):
  File "test.py", line 3, in <module>
    text = PapirusText(rotation = 360)
  File "/usr/local/lib/python2.7/dist-packages/papirus/text.py", line 14, in __init__
    self.papirus = Papirus(rotation = rotation)
  File "/usr/local/lib/python2.7/dist-packages/papirus/epd.py", line 58, in __init__
    self._lm75b = LM75B()
  File "/usr/local/lib/python2.7/dist-packages/papirus/lm75b.py", line 22, in __init__
    self._bus.write_byte_data(self._address, LM75B_CONF_REGISTER, LM75B_CONF_NORMAL)
IOError: [Errno 121] Remote I/O error

shawaj, 2017-08-12:
How did you install the software?

And you've definitely got the I2C and SPI turned on?

Could you try with one of the demos in bin folder as those are known to
work and if they won't then the setup hasn't worked properly

Thanks

Sharkeon, 2017-08-12:
I used the auto installation. I tried the manual installation, but every thing themes fine.

I turned I2c and SPI on.

I tried papirus-test but got this error:
Traceback (most recent call last):
  File "/usr/local/bin/papirus-test", line 68, in <module>
    papirus = Papirus()
  File "/usr/local/lib/python2.7/dist-packages/papirus/epd.py", line 58, in __init__
    self._lm75b = LM75B()
  File "/usr/local/lib/python2.7/dist-packages/papirus/lm75b.py", line 22, in __init__
    self._bus.write_byte_data(self._address, LM75B_CONF_REGISTER, LM75B_CONF_NORMAL)
IOError: [Errno 121] Remote I/O error

shawaj, 2017-08-12:
What Pi are you using this on?

You're not getting a connection to the temperature sensor by the looks of
it.

Can you try i2cdetect -y 1 ?

Can you also try starting from scratch on a clean version of Raspbian?

@tvoverbeek @francesco-vannini do you think we should maybe add an error
state where it'll still process the display commands with no temp sensor
but display an error message to screen saying "can't find temp sensor,
defaulting to 20 degrees Celsius" or something like that?

tvoverbeek, 2017-08-12:
First want to wait on the results from 'i2cdetect -y 1'.
Note if i2cdetect is not found do 'sudo apt-get install i2c-tools' first.
@Sharkeon what version of Raspbian are you running?
What is the output from 'uname -a' in a terminal?
Older versions of Raspbian did not have user pi in the i2c group -> hence no access to i2c unless running as root (with sudo).
@Sharkeon try running papirus-test as root: 'sudo papirus-test'. If that is working it is time to update your Raspbian.

Sharkeon, 2017-08-12:
I'm using a Raspberry pi zero with 2017-07-05-raspbian-jessie.

i2cdetect -y 1:
0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

uname -a : 
Linux zero 4.9.35+ #1014 Fri Jun 30 14:34:49 BST 2017 armv6l GNU/Linux

sudo papirus-test: 
Traceback (most recent call last):
  File "/usr/local/bin/papirus-test", line 68, in <module>
    papirus = Papirus()
  File "/usr/local/lib/python2.7/dist-packages/papirus/epd.py", line 58, in __init__
    self._lm75b = LM75B()
  File "/usr/local/lib/python2.7/dist-packages/papirus/lm75b.py", line 22, in __init__
    self._bus.write_byte_data(self._address, LM75B_CONF_REGISTER, LM75B_CONF_NORMAL)
IOError: [Errno 121] Remote I/O error

thanks for your help
i try to update my pi

shawaj, 2017-08-13:
What header have you soldered onto your Pi Zero?

tvoverbeek, 2017-08-13:
@Sharkeon  your display will not work with the Papirus library and drivers.
The Waveshare 2.13 epaper Display uses different GPIOs and a different comm protocol.
Follow the instructions on the Waveshare WiKi: http://www.waveshare.com/wiki/2.13inch_e-Paper_HAT
Closing this issue.

closed 2017-08-13 by tvoverbeek

shawaj, 2017-08-14:
Sorry @tvoverbeek I missed the bit about it being a wave share display! Good spot :-)

tvoverbeek, 2017-08-14:
Nothing to be sorry about. I also missed it initially.

Ton