PiSupply/PaPiRus #150: Looking to convert this to ruby

EddWills95, 2017-09-04:
Want to give the ruby crew some love on the rapsberry pi, i'm attempting to have crack at a port

I'm stuck trying to convert an image to the bytecode that gets put inside the `/dev/epd/LE/display_inverse` file
If anyone knows of an effective way to do this in ruby it'd be great to get me on the way. Been struggling for a few days now. 
`File.readbin` doesn't spit out the right thing for the screen to read it.

I realise this is a python library but figured i'd ask anyway.

shawaj, 2017-09-05:
So you're trying to port the whole thing including the code at repaper/gratis repo to Ruby or just this python library?

EddWills95, 2017-09-05:
Just the python library. I think i'd go a bit mad trying to do the whole EPD library ;) I just think it'd be really cool to integrate this into a small rails app on the pi. 

I feel like i'm close.
I can manipulate the image using MiniMagick (change to greyscale, bitdepth and the size). Save that file and then read it back in (not elegant at the moment. would just like to see it working)
`File.binread(img_file)`
gets me in the ball park of hexadecimal bytes. Then its just a question of writing those bytes to the file and then sending the command file the "U" Command. 

I've had some luck with this but for some reason ruby scrambles some of the data when it comes in and while it does get the screen to print something its just random dots all over the place. 

an extract of the File.binread: 

`#&')*)\x19\x1F-0-(0%()(\xFF\xDB\x00C\x01\a\a\a\n\b\n\x13\n\n\x13(\x1A\x16\x1A((((((((((((((((((((((((((((((((((((((((((((((((((\xFF\xC0\x00\`

So i'm not too sure whats going on here.

shawaj, 2017-09-06:
@tvoverbeek or @francesco-vannini do you have any tips? Since you have both
been playing with PaPiRus a lot more than me recently.

mmolhoek, 2017-10-01:
Hi @EddWills95, can you share the code you have untill now? I might have a look at it to see if I can make it work. i cloned your project [papiruby](https://github.com/EddWills95/PaPiRuby), but it seems to be an empty shell.

EddWills95, 2017-10-01:
checkout the dev branch i’ve set up.

The problem I was having was converting a picture into proper Hex Bytecode and then writing that to the appropriate file. 

Then calling the update function. 

I got it to actually put some black dots onto the screen but much beyond that couldn’t get it to work. The byte code always seems scrambled no matter how I tried to call it. 

Best of luck and if you have any progress I’ll see what I can do to help. 
Edd.

mmolhoek, 2017-10-01:
yes, a saw the dev branch after commenting :), so removed my comment, but you saw it anyway :)

in the coming days/weeks I will be busy with my own PaPiRus (got the 2.0 and 2.7" screen)

As I am more accustomed to ruby then python, I would love to see a ruby library for controlling the PaPiRus.

Anyway, I'll create PR's if i have something to contribute :)

EddWills95, 2017-10-01:
Nice one. All the best mate :D

shawaj, 2017-10-01:
@EddWills95 @mmolhoek FYI, I have now added this ruby repo to the readme - https://github.com/PiSupply/PaPiRus/blob/master/README.md

EddWills95, 2017-10-01:
Brilliant. Hopefully we can get somewhere with it and take some steps forward with the Ruby RasPi community :) 

Edd.

mmolhoek, 2017-10-02:
HI Aaron,

I think it is a bit to soon to add it to the library, as it is not working
yet.
I think people expect something working, if you put it in the readme :)

anyway, I am happy to join/help :) lets see if we can get it somewhere,
but don't get your hope up to much as I have little time to spent and many
projects to spend it on :)

but who knows...

Mischa

shawaj, 2017-10-02:
The reason for putting it in the readme is also so others could see it and
maybe provide some help :-)

mmolhoek, 2017-10-04:
@EddWills95, 
I have made a version that works using the [chunky_png](https://github.com/wvanbergen/chunky_png) library.
Could you create a chunky_png branch on your repo? then I will create a PR with this setup.
Please explain in the README of master that people can hop to the chunky_png branch to see the display in action :)

I still want to play with a version that uses Image Magick, as chunky_png has limited drawing tools,  but that is for another day.
At least now there is something working available that other people can play with.

I commented the code, so you should be able to see what's going on.

@shawaj , 👍 .

ps: Special thanks to @wvanbergen of chunky_png for his help with the bit shifting 👍

EddWills95, 2017-10-04:
Oh Fantastic ! 

https://github.com/EddWills95/PaPiRuby/tree/chunky_png

Theres the branch so throw a merge up there :)