PiSupply/PaPiRus #152: Waveshare e-ink displays support

JensC, 2017-09-09:
Hi All,

I wonder if someone plans to (or already works on) support the waveshare e-ink displays with PaPiRus, e.g the 2.9 and 1.54 inch displays. What is needed to get this working? Is it only to extend the epd_fuse driver or is there more to it?

Thanks in advance

shawaj, 2017-09-09:
I believe the connection cable is different so it'll be a hardware and
software change.

We're already working on support for similar displays though.

Do you already have the waveshare displays? Could you send pictures of
them? And links?

Thanks,
Aaron

tvoverbeek, 2017-09-10:
See e.g. http://www.waveshare.com/catalogsearch/result/?q=e-ink+display.
Waveshare is located in Shenzhen.
It seems most of their e-ink displays are 3-color (B/W/Red) and using an integrated timing controller (with Look Up Tables).
See their wiki for more development information for RPi and Arduino.

JensC, 2017-09-10:
@

JensC, 2017-09-10:
@shawaj yes I have both displays. They are pretty inexpensive (between 15 and 20 € at ebay Germany). So I could test any PaPiRus software to see how far it would work. @tvoverbeek yes I know their sources (wiki and libraries). However the waveshare software is, lets say it politely, not so great. I think supporting those displays would be of great benifit for a lot of iot developers.

shawaj, 2017-09-10:
@JensC our software won't work with these displays and also the displays won't connect into our driver boards so it won't work from both aspects

closed 2017-11-17 by francesco-vannini