PiSupply/PaPiRus #153: [feature request] add partial update flag to cli papirus-write

mikegleasonjr, 2017-09-15:


mikegleasonjr, 2017-09-15:
And papirus-draw!

shawaj, 2017-09-15:
@mikegleasonjr what exactly is the use case for this? how would you want it to work?

mikegleasonjr, 2017-09-15:
If I understand the device correctly, the library always write a full image to the device right? Even if you write text, it is converted as an image and sent to the device.

I already have a program that generates images. Like pipes on Linux, instead of writing a program that links to the library, I would just produce the images on my end and use the existing cli to send them to the device.

shawaj, 2017-09-15:
but what are you meaning with the partial updates? as you can already do that?

mikegleasonjr, 2017-09-15:
If I use the library directly. I can do it. I just want the CLIs to support it.

I want to periodically do (oversimplified)...

```
./papirus-draw my-image-1.bmp
sleep(5)
./papirus-draw --partial my-image-2.bmp
sleep(5)
./papirus-draw --partial my-image-3.bmp
sleep(5)
./papirus-draw --partial my-image-4.bmp
[...]
```

shawaj, 2017-09-15:
@francesco-vannini @tvoverbeek any thoughts on this?

tvoverbeek, 2017-09-15:
If I look at @mikegleasonjr 's example above then I suggest to look at papirus-animation which @francesco-vannini added today. It does what what @mikegleasonjr wants with a single call to papirus-animation: 'papirus-animation --delay 5 my-image-1.bmp'
Works best when the .bmp already are monochrome, so dithering is a no-op.
When color/grayscale pictures are dithered, partial update does not work very well (lots of ghosting).

shawaj, 2017-09-15:
I think the idea though was to have a --partial tag on the standard papirus-draw and papirus-text command line tools, in order to force partial updates for them.

I had a feeling we already had this in some way or another due to work from @BenScarb ? Might be wrong though!

mikegleasonjr, 2017-09-15:
@tvoverbeek wow this is great, I could use that.

(I don't know how to say it in Unix terms, named pipes or sockets but) I could even use a single file descriptor opened by one of my program, each read by `papirus-animation` on this file would make my program write an updated version of an image. Since `papirus-animation` works in loops, this would work.

@francesco-vannini talk about timing!

Thanks @shawaj for raising the discussion.

closed 2017-09-15 by mikegleasonjr

francesco-vannini, 2017-09-15:
Glad it works out for you @mikegleasonjr , yeah the timing is quite scary ;)

mikegleasonjr, 2017-09-18:
@shawaj @tvoverbeek @francesco-vannini

Here are some rendering of my program:

<kbd>[image]</kbd>

<kbd>[image]</kbd>

<kbd>[image]</kbd>

 Will be publishing on github once I receive my Pi and my screen.

I was able to develop it without hardware because...

* I use named pipes (FIFO) to write my images to (every 5 seconds)
  * I will then leverage `papirus-animation` to display the generated image
* I use snmp to gather statistics
  * You will be able to use the Pi ou point it to your router

That's it

shawaj, 2017-09-18:
Looks really cool. Definitely upload the code.

Are you going to add it to this main repo as am example?

mikegleasonjr, 2017-09-18:
@shawaj I wrote it in Go so I guess I would put it on my repo and if you guys like it you could link it. If you guys want it here I can also donate it here. Let me receive my Pi and test on it first :D

shawaj, 2017-09-18:
Yeah great, stick it in your repo and we will link to it!