PiSupply/PaPiRus #158: switches not working with GPIOZero

vwillcox, 2017-09-26:
Currently I can't call the switches with GPIOZero. They do not respond.

shawaj, 2017-09-26:
Have you soldered them in?

There's no real way they can't work. What gpio are you trying to use?

vwillcox, 2017-09-27:
I can use the buttons with normal GPIO 

I converted:

      SW1 = 16
       SW2 = 26
       SW3 = 20
       SW4 = 21

into 

sw1 = Button(16)
sw2 = Button(26)
etc

Then ran a button_ispressed() and got nothing out

francesco-vannini, 2017-09-27:
Hello,

Try with this approach instead

from gpiozero import Button
sw1 = Button(16)
sw1.wait_for_press()
print("SW1 was pressed!")

vwillcox, 2017-09-27:
pi@raspberrypi:~/speedtest-cron $ python
Python 2.7.13 (default, Jan 19 2017, 14:48:08)
[GCC 6.3.0 20170124] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from gpiozero import Button
>>> sw1 = Button(16)
>>> sw1.wait_for_press()

nothing....

tvoverbeek, 2017-09-27:
When looking at the schematics for the PaPiRus HAT ([https://github.com/PiSupply/PaPiRus/blob/master/hardware/PaPiRus%20HAT/Latest%20Version%20-%20v1.9/2014-035-01-Pi-ePaper-circuit_v1_9.pdf](url)) you see that the switches have their own pull-up resistors on the board. When using the pull_up=False argument to Button, there is an RPi internal pull_down to ground (ca 50k). When SW1 is open the gpio input will be high, when closed low. This corresponds to button_pressed and button_released respectively, i.e. reversed logic.

Change your code to the following:
```
from gpiozero import Button
sw1 = Button(16, pull_up=False)
sw1.wait_for_release()
print('Sw1 pressed')
```

vwillcox, 2017-09-27:
Thanks :) I see that does now work. I believe this is closed :)

closed 2017-09-27 by vwillcox