PiSupply/PaPiRus #161: No output - hangs at "Writing to Papirus...."

tobiasvogel, 2017-10-14:
I just compiled from the git-master https://github.com/PiSupply/PaPiRus/commit/b8c3619b62fb8ed854c6950dea6661302e80580b on a Raspberry Pi 3 with Raspbian (Kernel 4.9.35-v7+) and everything looked ok so far.

However, any of the included demos hang at the output "Writing to Papirus...." and nothing ever happens from there.

The script-process can neither be interrupted by Ctrl+C nor killed (even using kill -9).
Interrupting by stopping the epd-fuse service works, though.

shawaj, 2017-10-14:
Sounds like a bad connection or a broken display.

What happens if you run sudo i2cdetect -y 1 ?

I just compiled from the git-master b8c3619

tobiasvogel, 2017-10-14:
Output:

`$ sudo i2cdetect -y 1`
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- UU 
70: -- -- -- -- -- -- -- --
```

tvoverbeek, 2017-10-14:
Here is the expected output in case of the HAT:

>      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
> 00:          -- -- -- -- -- -- -- -- -- -- -- -- --
> 10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- --
> 50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- UU
> 70: -- -- -- -- -- -- -- --

0x48 is the LM75B temperature sensor on both the HAT and Zero
0x6F is the Real Time Clock, only on the HAT.
UU means you have a kernel module taking care of the i2c
device on 0x6F.
For the HAT it would mean you have a 'dtoverlay=i2c-rtc,mcp7940x=1'
in /boot/config.txt.
If you have an other device on 0x6F and the HAT then you have a conflict
which might explain the hang you are seeing.
Anyway, the missing 0x48 is suspect, either bad contacts, conflict on the
i2c bus
or a faulty Papirus.

tobiasvogel, 2017-10-14:
I do indeed have the line 'dtoverlay=i2c-rtc,mcp7940x=1' in /boot/config.txt
and on commenting this line out, the "UU" turns into 0x6f upon reboot.
There are no other devices (except for a Raspberry Pi Camera Module V2) connected
to my Raspberry Pi.
I noticed there is no /dev/rtc device, though. Would I need to create such a device node
myself or is the Kernel module supposed to do that automatically?

tvoverbeek, 2017-10-14:
The 'no /dev/rtc' is expected when you do not have the kernel module loaded.
The HAT demos in /usr/local/bin all work without the kernel module loaded.
0x6f (and the 'UU' before) mean it sees the RTC on the HAT, but the temperature
sensor (LM75B) is not detected on the i2c bus.
Smells like a bad board then. Suggest you contact the supplier for a replacement.

shawaj, 2017-10-17:
@francesco-vannini do you think this could be related to the issues you have been testing today?

francesco-vannini, 2017-11-17:
@tobiasvogel has this been fixed after the latest commit from @tvoverbeek ?