PiSupply/PaPiRus #162: Software caused connection abort followed by no such file /dev/epd/version

GabiSchaffzin, 2017-10-17:
Hi,

I've followed the instructions for both [software](https://github.com/PiSupply/PaPiRus) and [hardware](https://www.pi-supply.com/make/papirus-assembly-tips-and-gotchas/), I've checked out [Frederick Vandenbosch's post](http://frederickvandenbosch.be/?p=1483), as well as Issues [3](https://github.com/PiSupply/PaPiRus/issues/3) and [64](https://github.com/PiSupply/PaPiRus/issues/64) but am still running into the same problem:

1. Install SW and HW as instructed including pogo and buttons.
2. `curl -sSL https://pisupp.ly/papiruscode | sudo bash`
3. Select 2.7" screen
4. run `papirus-clock` (or `papirus-test` or `papirus-write "hello world!"`
5. Results in 
```
panel = EPD 2.7 264 x 176  version=4 COG=2 FILM=231
...
IOError: [Errno 103] Software caused connection abort
```
6. run command again
7. Results in `IOError: [Errno 2] No such file or directory: '/dev/epd/version'`

I've tried manually setting I2C and SPI via `raspi-config`, I've tried manually setting the screen size via `papirus-config`.

When I run `sudo i2cdetect -y 1` (as suggested in Issue [113](https://github.com/PiSupply/PaPiRus/issues/113), results:

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 6f
70: -- -- -- -- -- -- -- --
```

When I re-run install or config, then `papirus-clock`, the first error is
`IOError: [Errno 103] Software caused connection abort`
Running `papirus-clock` again results in
`IOError: [Errno 2] No such file or directory: '/dev/epd/version'`

Can you help me out here?

shawaj, 2017-10-17:
@tvoverbeek @francesco-vannini do you think this could be an issue due to gratis code not working or a recent Raspbian update causing a conflict? Seems like a similar issue to the other ones we have seen.

@GabiSchaffzin - is this PaPiRus HAT or Zero?

tvoverbeek, 2017-10-18:
Must be the Papirus HAT (2.7" screen and pogo pin.
Yes this is the same issue as @francesco-vannini  experienced.
It seems to occur on RPis without built-in WiFi and Bluetooth (= all models except Pi 3 and Pi Zero W).
The epd-fuse service crashes on the first use of the full update command.
I have experienced the same issue now on a Pi Zero V1.2.
@GabiSchaffzin when epd-fuse crashes the service shutdown removes /dev/epd.
As a work-around for now restart the service manually: 'sudo systemctl start epd-fuse'.
After manual restart it works  again.
@GabiSchaffzin Is this with the full Raspbian Stretch or Stretch-lite? and what model RPi?
Will be investigating the cause of the crash in the next few days.

GabiSchaffzin, 2017-10-18:
Thanks @shawaj and @tvoverbeek for the quick input. Apologies for the delay.

This is an Raspberry Pi 3 - Model B - ARMv8 with 1G RAM with Raspbian Stretch-lite.

Will be at my studio later today and will try restarting epd-fuse.

GabiSchaffzin, 2017-10-19:
Updating here to say that I wiped my SD and reinstalled Raspiban from scratch—the full Stretch desktop version, just in case the Lite version was causing problems.

Unfortunately, I'm still getting the same issue:

```
$ papirus-test
Assuming this is a PaPiRus HAT board

---- Detect temperature sensor and RTC (RTC for HAT only) ----
NXP LM75BD detected at: 0x48
MCP7940N detected at: 0x6f

---- Screen Information ----
Vendor: Pi Supply
Product: PaPiRus ePaper HAT v1.9
Serial: 342d3832-6464-2d35-3561-32636536327b
Date/Time from RTC: Monday 01 January 2001 - 00:00:00
Temperature from sensor: 35 degrees Celsius
Panel info: EPD 2.7 264 x 176 version=4 COG=2 FILM=231

Traceback (most recent call last):
  File "/usr/local/bin/papirus-test", line 285, in <module>
    main(sys.argv[1:])
  File "/usr/local/bin/papirus-test", line 84, in main
    if not break_detect(papirus): # trying to use the screen
  File "/usr/local/bin/papirus-test", line 203, in break_detect
    papirus.update()
  File "/usr/local/lib/python2.7/dist-packages/papirus/epd.py", line 201, in update
    self._command('U')
  File "/usr/local/lib/python2.7/dist-packages/papirus/epd.py", line 217, in _command
    f.write(b(c))
IOError: [Errno 103] Software caused connection abort
```

Considering I'm getting a proper temperature reading, I'm assuming this isn't necessarily a GPIO hardware issue?

shawaj, 2017-10-19:
@tvoverbeek has worked out what the issue is and a fix is being investigated. Thanks for your patience

tvoverbeek, 2017-10-19:
@GabiSchaffzin Crash is caused by multiple initializations (by epd-fuse and by the system) of the serial GPIOs 14 and 15. If system comes first the epd-fuse initialization fails and a pointer used to the general EPD info does not get set. Any following command (clear, update, display, ...) is referencing this structure and causes a SIGSEV (memory access violation) and epd-fuse aborts.

Can you try the following:
in file /usr/lib/systemd/system/epd-fuse.service change line 3 from
`After=network.target` to
`After=network.target dev-serial1.device`
and reboot
This should ensure epd-fuse initialization runs after the system initialization of the serial ports.
You can check with `gpio readall` if it worked.
Expected output for BCM 14 and 15:

``
pi@rpi-stretch-full:~ $ gpio readall
 +-----+-----+---------+------+---+---Pi 3---+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 | ALT0 | 1 |  3 || 4  |   |      | 5v      |     |     |
 |   3 |   9 |   SCL.1 | ALT0 | 1 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 1 |  7 || 8  | 0 | OUT  | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 0 | OUT  | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |   IN | 1 | 11 || 12 | 0 | IN   | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   IN | 0 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   IN | 0 | 15 || 16 | 0 | OUT  | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 | OUT  | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI | ALT0 | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO | ALT0 | 0 | 21 || 22 | 0 | IN   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK | ALT0 | 0 | 23 || 24 | 1 | OUT  | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 1 | OUT  | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   IN | 1 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |   IN | 1 | 31 || 32 | 0 | IN   | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 |   IN | 1 | 35 || 36 | 1 | IN   | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   IN | 1 | 37 || 38 | 1 | IN   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 1 | IN   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+---Pi 3---+---+------+---------+-----+-----+
pi@rpi-stretch-full:~ $ 
``

GabiSchaffzin, 2017-10-19:
@tvoverbeek thank you very much for looking into this and the explanation.

I made the change you recommended and am still receiving the error.

`gpio readall` output:

```
+-----+-----+---------+------+---+---Pi 3---+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 | ALT0 | 1 |  3 || 4  |   |      | 5v      |     |     |
 |   3 |   9 |   SCL.1 | ALT0 | 1 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 1 |  7 || 8  | 0 | IN   | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 1 | IN   | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |   IN | 1 | 11 || 12 | 0 | ALT0 | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   IN | 1 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   IN | 0 | 15 || 16 | 0 | IN   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 | IN   | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI | ALT0 | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO | ALT0 | 0 | 21 || 22 | 0 | IN   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK | ALT0 | 0 | 23 || 24 | 0 | OUT  | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 1 | OUT  | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   IN | 1 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |   IN | 1 | 31 || 32 | 0 | IN   | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 | ALT0 | 0 | 35 || 36 | 1 | IN   | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   IN | 1 | 37 || 38 | 1 | IN   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 1 | IN   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+---Pi 3---+---+------+---------+-----+-----+
```

tvoverbeek, 2017-10-19:
It seems epd-fuse did not initialize anything.
What is the output of `systemctl status epd-fuse`?
I expect a failed state.
Something fishy is pin 35 (BCM 19) which is in state ALT0.
This is used as the frame sync when you use I2S sound.
Any other hardware enabled in /boot/config.txt (dtoverlay=???).

Have you tried restarting the service (`sudo systemctl start epd-fuse`)?
Does that succeed?

GabiSchaffzin, 2017-10-19:
Yep, that was it.

I am using an external HDMI touch-screen, not the one sold through rpi.org. It requires extra code in boot.txt, including a dtoverlay entry.

I removed that entry and the screen is working like a charm.

My sincere apologies for sending you on this goose-chase and it being unrelated to the product. **Thank you** for the help!

closed 2017-10-19 by GabiSchaffzin

GabiSchaffzin, 2017-10-19:
Quick update for those who find this thread via Google.

The screen I'm using is the `Elecrow RPA05010R HDMI 5-Inch 800x480 TFT LCD Display with Touch Screen Monitor for Raspberry Pi B+/2B/3B`

You can still use it without the touchscreen capabilities, just the display by only commenting out
`dtoverlay=ads7846,cs=1,penirq=25,penirq_pull=2,speed=50000,keep_vref_on=0,swapxy=0,pmax=255,xohms=150,xmin=200,xmax=3900,ymin=200,ymax=3900` 
in `/boot/config.txt`

tvoverbeek, 2017-10-20:
Just for the sake of completeness Papirus uses the following GPIOs (Broadcom numbering):
2,3 (I2C)
8,9,10,11 (SPI0)
14,15,23,24,25 (control signals for display)
The conflict was with the penirq  (25) of the touchscreen.