PiSupply/PaPiRus #164: PaPiRus Zero 5volt supply

uselesshippy, 2017-10-31:
The PaPiRus Zero pHat has a 5v supply listed, but as far as I can see from the schematics the 5v line is not used.

Can I omit this connection and still have the full functionality of the pHat?

I ask this as I want to drive the screen from an Arduino, and I only have 3.3v available.

Thanks

tvoverbeek, 2017-10-31:
Should work. The 5V is only connected to the solder pad on the pHAT which normally is not used.
Waiting for Aaron or Francesco from Pi SUpply to confirm.

shawaj, 2017-10-31:
You can see the datasheet here - https://github.com/PiSupply/PaPiRus/blob/master/hardware/PaPiRus%20Zero/Latest%20Version%20-%20v1.2/2015-032-01-PaPiRus-ZERO-circuit_v1_2.pdf

As you can see - the 5v is not used other than to breakout the 5v to the FPC extension connector on the PCB and for the 5v pad.

Hope that helps!

closed 2017-10-31 by shawaj

uselesshippy, 2017-10-31:
That's what I thought, but I just wanted to check.

Thank you.