PiSupply/PaPiRus #166: add info about papirus dtoverlay

shawaj, 2017-12-23:
see here - https://github.com/raspberrypi/linux/blob/rpi-4.9.y/arch/arm/boot/dts/overlays/README

there is a papirus dtoverlay which needs documenting

tvoverbeek, 2017-12-23:
As a preliminary start here the info from Noralf Trønnes (the author of the kernel module,
also well known for his support for TFT displays)
```
From: "Noralf Trønnes" <notifications@github.com>
Date: 22 Jul 2017 6:15 pm
Subject: Re: [repaper/gratis] Linux kernel driver - re-license code under GPLv2? (#68)
To: "repaper/gratis" <gratis@noreply.github.com>
Cc: "Aaron Shaw" <shawaj@gmail.com>, "Mention" <mention@noreply.github.com>

The driver is now merged in the official rpi kernel and available through rpi-update. It will later be available through the debian kernel package and probably the next Raspian release.

Pull request: raspberrypi/linux#2119
Driver: drivers/gpu/drm/tinydrm/repaper.c
DT overlay: https://github.com/raspberrypi/linux/blob/rpi-4.9.y/arch/arm/boot/dts/overlays/papirus-overlay.dts

Here's the DT overlay README entry:

Name:   papirus
Info:   PaPiRus ePaper Screen by Pi Supply (both HAT and pHAT)
Load:   dtoverlay=papirus,<param>=<val>
Params: panel                   Display panel (required):
                                1.44": e1144cs021
                                2.0":  e2200cs021
                                2.7":  e2271cs021

        speed                   Display SPI bus speed
Use one these in /boot/config.txt:

dtoverlay=papirus,panel=e1144cs021
dtoverlay=papirus,panel=e2200cs021
dtoverlay=papirus,panel=e2271cs021
The driver presents the framebuffer as XRGB8888 to userpace and converts it to monochrome using ITU BT.601. This way it can be used with all graphics libraries. See tinydrm_xrgb8888_to_gray8() in drivers/gpu/drm/tinydrm/core/tinydrm-helpers.c.

I've halfmade a library pydrm that might be useful with these displays, time will tell.
```
See also the discussion in Issue #68 in the repaper/gratis repository.
Note: using the kernel module is mutually exclusive with using the epd-fuse driver.

shawaj, 2017-12-23:
Thanks @tvoverbeek

Have you tried this out at all?

tvoverbeek, 2017-12-23:
@shawaj Yes, last July.
```
The kernel module makes the Papirus available as /dev/fb1.
Using nitro's instructions from https://github.com/notro/fbtft/wiki/Boot-console you get part
of the boot message on the epaper display.
I changed the console font to 6x12.
```
[image]

shawaj, 2017-12-23:
Looks really cool. Can it mirror framebuffer to the display itself or only
the boot message / console?

tvoverbeek, 2017-12-29:
@shawaj In principle yes, but the drm code maps 24-bit color to only black and white without dithering.
You could even start a separate X-windows desktop on the epaper display, but it also suffers from the color to pure black and white mapping.
So pretty much useless for the desktop.

Since the update rate for epaper is slow, a simple grab of the color desktop framebuffer, then converting it to a dithered B/W image fitting the epaper display and displaying it in a loop (similar to papirus-cam) works fine.
Have a version working which uses the dtoverlay.
Next is to convert it to using the epd-fuse driver (and add it to the repository)

shawaj, 2017-12-29:
Oh awesome - that sounds really cool :-)

shawaj, 2018-04-07:
and maybe we can close this one too also? based on https://github.com/PiSupply/PaPiRus/blob/master/bin/papirus-fbcopy

although maybe still need to document the overlay?

tvoverbeek, 2018-04-07:
Leave it open, till we have documented the overlay.
The overlay is mutually exclusive with our 'normal' driver, but is useful for certain use cases
(e.g. shell scripts producing info on a terminal).