PiSupply/PaPiRus #167: Potential to configure buttons with an overlay?

shawaj, 2017-12-25:
Similar to here - https://github.com/notro/fbtft/blob/master/dts/overlays/rpi/tinylcd35-overlay.dts

shawaj, 2018-01-03:
also see here - https://github.com/shawaj/linux/blob/a7eca81e0c579885f075549e19abae42408065b6/arch/arm/boot/dts/overlays/gpio-key-overlay.dts

and here - https://github.com/raspberrypi/linux/pull/2329

This has now been implemented in the core kernel. You can call multiple buttons in the `/boot/config.txt` file using multiple calls to the overlay:
```
dtoverlay=gpio-key,gpio=13,keycode=59,label="KEY_F1"
dtoverlay=gpio-key,gpio=17,keycode=60,label="KEY_F2"
dtoverlay=gpio-key,gpio=22,keycode=61,label="KEY_F3"
dtoverlay=gpio-key,gpio=26,keycode=62,label="KEY_F4"
dtoverlay=gpio-key,gpio=27,keycode=63,label="KEY_F5"
```

Or on the command line at runtime like:
```
sudo dtoverlay gpio-key gpio=21 keycode=4 label="KEY_4" gpio_pull=0
```

Again, multiple calls can be done to this.

@francesco-vannini do you think we should add info about how to set buttons up like this to the README / docs somewhere? Could be pretty useful for navigating menus potentially?

kog8790, 2021-02-23:
This is super helpful, thanks!