PiSupply/PaPiRus #168: papirus-write parsing line break \n

jarrah31, 2018-01-03:
I know that PapirusTextPos will take into account \n as a line break using the following example:

`text.AddText("hello\nworld", 10, 20, Id="Start" )`

However if I use papirus-write, \n just appears within the text on the display.

papirus-write.py "hello\nWorld"

Is there a workaround by any chance?

Thanks.

tvoverbeek, 2018-01-03:
The '\n' is not converted by the shell (bash).
This works:
`papirus-write $'hello\nWorld'`
See e.g. https://stackoverflow.com/questions/3005963/how-can-i-have-a-newline-in-a-string-in-sh
Note: you need to use single quotes after the '$', double quotes do not work.

closed 2018-01-03 by tvoverbeek

jarrah31, 2018-01-04:
Ahh of course, thank you!

shawaj, 2018-01-04:
@jarrah31 @tvoverbeek added to the main readme now for reference :-)