PiSupply/PaPiRus #172: Partial vs Full Refresh Rate and Power Down Question

jarrah31, 2018-01-08:
I'm sorry as this isn't an issue as such, but not sure where else to ask the question.

Regarding the sentence below, when you say "completely power down the screen every few hours", does this mean shutting the Pi down and removing the power? This wouldn't be very practical for long-term use, so what happens if I don't do that? 

> "You should refresh the screen using a Full Update every few minutes and it is also recommended to completely power down the screen every few hours."

I'm planning to use the display to show outside temperature which will partial update every 5 minutes. Is there a general guide or ratio to indicate how often a full refresh is needed after X many partial updates? For example, I see the clock program does a full update every 60 partial updates, so would I do a full refresh every 300 minutes? 

Thanks in advance.

shawaj, 2018-01-08:
Is there a reason to do a partial update every 5 minutes instead of a full update?

It's better to do a full update unless your use case specifically can't (needs quicker updates).

The full per down thing isn't essential it can just elongate the life of the display. If you're only updating every five minutes and doing a full update you should be OK

jarrah31, 2018-01-09:
I'm using the display to show Home Assistant sensor status updates, and as it's in line of site on my desk, the full updates (going black to white) can be a little distracting. My thought for the partial updates would mean a neater, less noticeable change of the numbers.

I'm also likely to change how often it updates as I work out ways to show more sensors, so for example a partial update would show how many lights are on and may be updated as often as every couple of seconds to many tens of minutes. I was thinking of keeping a count of partial updates so that a full one occurs every X partials.

How will you know when the display life is coming to an end out of interest?

On the Home Assistant side of things - it's a great home automation product that runs on a Raspberry Pi, and would love it if Papirus could be integrated as a component on there! :) https://home-assistant.io

shawaj, 2018-01-09:
Fair enough. As said it's better to do a full update but if you want to use partial that's also fine. Just make sure you so a full update as often as possible really.

And there's no real way to tell that to be honest it either works or it doesn't.

About home assistant, maybe submit an issue to them asking if they can integrate it. We'd be happy to help.

jarrah31, 2018-01-23:
Thanks for your help - and yes Home Assistant integration would be fab! At the moment I'm trying to write code that uses HA's API which seems to be working well for now.

Quick question - can partial screen updating be done with positional text?

For example, if I used "text = PapirusTextPos", is there a way to only partially update blocks of text using ID or can I only do text.WriteAll()?

tvoverbeek, 2018-01-23:
The answer is yes, but you still have to use textWriteAll(). I'll explain.

The image to be displayed is prepared in RPi memory before it is sent to the display.
When using PapirusTextPos you can update part of the RPi memory image by specifying the ID.
The complete image is sent to the display.
When you use partial update the chip on the display only updates the pixels which have changed from the previous image.
You can specify to use partial instead of full update to textWriteAll: `text.WriteAll(partialUpdate=True)`.
I believe this combination does what you want.

jarrah31, 2018-01-23:
Fantastic, just what I was after and it works a treat, thank you! 

It would be great if this could be added to the guide on the main page?

tvoverbeek, 2018-01-23:
It is already there in the main README.
See e.g. "The Positional Text API (example 2)" and in the Notes after example 3:
```
When using the PapirusTextPos, in either mode, setting the "partialUpdates" property to
True will cause partial updates to be done, meaning only the section of the PaPiRus screen
that has been changed will be updated. These can be vastly quicker than a full update
for each piece of text.

If not using the "partialUpdates" property, calling WriteAll(True) will do the same thing
on a one off basis.
```

VirtualWolf, 2018-02-11:
I was about to raise this exact question myself!

> The full per down thing isn't essential it can just elongate the life of the display.

I'm planning on doing basically exactly the same thing as @jarrah31, using the PaPiRus to display temperatures and updating every five minutes or so. What sort of display life would you typically expect from this pattern of usage?

shawaj, 2018-02-12:
You mean doing a single full refresh every 5 minutes?

VirtualWolf, 2018-02-12:
Yep, exactly.