PiSupply/PaPiRus #175: Issues setting up Papirus on Pi Zero W

gtrez, 2018-01-18:
Hi guys

Apologies in advance for my limited lack of knowledge. I have experience using pi's for various home networking projects but not surrounding issues like I am having.

I installed Raspbian Lite on a Raspberry Pi Zero W with the Papirus 2" screen attached.
Set up networking, enabled the relevant interfacing options via raspi-config.
Tried the auto install of papirus using curl -sSL https://pisupp.ly/papiruscode | sudo bash
The pi appears to install everything succesfully until I choose the screen size "2.0". 

I then get a double error (both the same message):
**2.0
A dependency job for epd-fuse.service failed. See 'journalctl -xe' for details.
A dependency job for epd-fuse.service failed. See 'journalctl -xe' for details**

The pi restarts and I can no longer SSH in. When I plug the HDMI in and view the boot on my TV I am getting lines and lines of random text scrolling so fast I can't really make out what it is saying (possible memory dump?). This never ends. 

Where have I gone wrong?

Many thanks in advance.

shawaj, 2018-01-18:
Have you tried on a full instance if Raspbian instead of lite?

@tvoverbeek @francesco-vannini have you ever tried with Raspbian Lite? I can't say I ever have

gtrez, 2018-01-18:
I haven't because I bought a 4gb micro SD specifically because I didn't think I'd need full Raspbian, but this did cross my mind. If this is a requirement I'll order a new SD card and give it a go. If anyone has any experience with Raspbian Lite and may have an idea of how to resolve this issue that would be great, if not I'll get a new card and give it another go!

Many thanks for the very speedy response, I was not expecting that!

tvoverbeek, 2018-01-18:
For me it works fine with Raspbian Lite. Which Raspbian are you using? It should be either Jessie or Stretch, preferably Stretch. My smallest SD cards nowadays are 8GB, but 4GB should be fine for Raspbian lite. Does it work fine before installing Papirus?

gtrez, 2018-01-18:
It works great before installing Papirus. Like I said, everything is all set up, ssh is working and nothing seems to be amiss. I am running version 9 stretch.

I have just tried a fresh install of Raspbian Lite, and I am attempting the manual install to see if I can pinpoint the exact cause. I'm getting the following errors after cd /tmp/papirus/gratis

pi@raspberrypi:/tmp/papirus/gratis $ make rpi EPD_IO=epd_io.h PANEL_VERSION='V231_G2'
make DESTDIR="" PREFIX=/usr SERVICE=systemd PLATFORM=../RaspberryPi PANEL_VERSION="V231_G2" EPD_IO="epd_io.h" -C PlatformWithOS/driver-common
make[1]: Entering directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"'   -c -o gpio_test.o gpio_test.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"'   -c -o gpio.o ../RaspberryPi/gpio.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"' -lfuse -pthread -lrt -o "gpio_test" gpio_test.o gpio.o
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"'   -c -o epd_test.o epd_test.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"'   -c -o spi.o spi.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"'   -c -o epd.o V231_G2/epd.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"' -lfuse -pthread -lrt -o "epd_test" epd_test.o gpio.o spi.o epd.o
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"'   -c -o epd_fuse.o epd_fuse.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"' -lfuse -pthread -lrt -o "epd_fuse" epd_fuse.o gpio.o spi.o epd.o
make[1]: Leaving directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
pi@raspberrypi:/tmp/papirus/gratis $ make rpi-install EPD_IO=epd_io.h PANEL_VERSION='V231_G2'
make DESTDIR="" PREFIX=/usr SERVICE=systemd PLATFORM=../RaspberryPi PANEL_VERSION="V231_G2" EPD_IO="epd_io.h" -C PlatformWithOS/driver-common install
make[1]: Entering directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
[ -e epd_fuse ] || false
[ ! -d "/usr/sbin" ] && mkdir -p "/usr/sbin" || true
install --group=root --mode=750 --owner=root epd_fuse "/usr/sbin"
install: cannot create regular file '/usr/sbin/epd_fuse': Permission denied
Makefile:54: recipe for target 'install' failed
make[1]: *** [install] Error 1
make[1]: Leaving directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
Makefile:87: recipe for target 'rpi-install' failed
make: *** [rpi-install] Error 2

tvoverbeek, 2018-01-19:
@gtrez epd_fuse built OK, but the install failed since you did not use `sudo`.
The last steps in the manual install should be:
```
sudo make rpi-install EPD_IO=epd_io.h PANEL_VERSION='V231_G2'
sudo systemctl enable epd-fuse.service
sudo systemctl start epd-fuse
sudo papirus-set 2.0
```

gtrez, 2018-01-19:
Followed instructions per above with sudo and after sudo systemctl start epd-fuse I get the same error message:
**A dependency job for epd-fuse.service failed. See 'journalctl -xe' for details.**

journalctl -xe gives:

Jan 19 18:02:12 raspberrypi sudo[742]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Jan 19 18:02:12 raspberrypi systemd[1]: Reloading.
Jan 19 18:02:13 raspberrypi systemd[1]: apt-daily.timer: Adding 5h 22min 28.562964s random time.
Jan 19 18:02:13 raspberrypi systemd[1]: apt-daily-upgrade.timer: Adding 42min 11.102627s random time.
Jan 19 18:02:13 raspberrypi sudo[742]: pam_unix(sudo:session): session closed for user root
Jan 19 18:02:19 raspberrypi sudo[760]:       pi : TTY=pts/0 ; PWD=/tmp/papirus/gratis ; USER=root ; COMMAND=/bin/systemctl start epd-fuse
Jan 19 18:02:19 raspberrypi sudo[760]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Jan 19 18:03:49 raspberrypi systemd[1]: dev-spidev0.0.device: Job dev-spidev0.0.device/start timed out.
Jan 19 18:03:49 raspberrypi systemd[1]: Timed out waiting for device dev-spidev0.0.device.
-- Subject: Unit dev-spidev0.0.device has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit dev-spidev0.0.device has failed.
-- 
-- The result is timeout.
Jan 19 18:03:49 raspberrypi systemd[1]: Dependency failed for A Fuse driver for the Repaper.org EPD.
-- Subject: Unit epd-fuse.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit epd-fuse.service has failed.
-- 
-- The result is dependency.
Jan 19 18:03:49 raspberrypi systemd[1]: epd-fuse.service: Job epd-fuse.service/start failed with result 'dependency'.
Jan 19 18:03:49 raspberrypi sudo[760]: pam_unix(sudo:session): session closed for user root
Jan 19 18:03:49 raspberrypi systemd[1]: dev-spidev0.0.device: Job dev-spidev0.0.device/start failed with result 'timeout'.

gtrez, 2018-01-19:
Sorry not sure why my previous post is in huge font...

gtrez, 2018-01-19:
Guys, I found the problem. Started fresh, took the whole thing apart. One of my pins was soldered very slightly higher than the rest. Resoldered, fresh install, did the auto setup and it works a treat. Thank you again for all your time, and sorry it was something so stupid.

closed 2018-01-19 by gtrez

tvoverbeek, 2018-01-19:
Glad you solved it.

gtrez, 2018-01-19:
Thanks again for your help :+1: