PiSupply/PaPiRus #176: The screen might be broken, cannot continue further

lexuv2, 2018-01-21:
I brought new papirus and it was working fine for a week, then i tried to set up https://github.com/f4exb/papiclock  ,and after `papirus-test` I got 
`Assuming this is a PaPiRus Zero board

---- Detect temperature sensor and RTC (RTC for HAT only) ----
NXP LM75BD detected at: 0x48

---- Screen Information ----
Temperature from sensor: 32 degrees Celsius
Panel info: EPD 2.0 200 x 96 version=4 COG=2 FILM=231

The screen might be broken, cannot continue further`
after `sudo i2cdetect -y 1` I got 
`   0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
`
I'm sure that screen is not broken 
using raspberry pi zero w
and I tried reinstalling raspbian

tvoverbeek, 2018-01-21:
What do you get as output for `cat /dev/epd/error` after running the failed papirus-test?
Looking at https://github.com/f4exb/papiclock this is running on the larger 2.7" display.
Are you trying it on a 200x96 (Pi zero size) display?
Anyway it detects the clockchip on the PaPiRus Zero board.

lexuv2, 2018-01-21:
I got Panel broken and I have 2 inch display 200x96

tvoverbeek, 2018-01-21:
You can also get this message if the flex cable from the display to the FPC connector on the board is not properly connected.

lexuv2, 2018-01-21:
[image]

I think i got the issue

tvoverbeek, 2018-01-21:
That does not look good.
Any idea how it got damaged?

lexuv2, 2018-01-21:
No it was glued to the base

tvoverbeek, 2018-01-21:
Probably time to contact the supplier. Did you get it from Pi Supply?

lexuv2, 2018-01-22:
by the double sided tape what was included in kit

lexuv2, 2018-01-22:
I think I was a delivery problem

shawaj, 2018-01-23:
If it was damaged like that on delivery, there's no way it would have worked in the first place. 

Anyway can you send a picture of the entire display too?

lexuv2, 2018-01-23:
[image]
[image]
I think it was damaged a little, and I made it bigger

shawaj, 2018-01-23:
You can purchase a new screen here - https://www.pi-supply.com/product/epaper-displays/