PiSupply/PaPiRus #182: No temperature sensor at 0x48

lawik, 2018-03-24:
Hi

I've an issue when running the tests. papirus-test fails at the animation (partial update) step. The screen is showing some parts of graphics and stuff.

i2cdetect -y 1 gives a 0x6f but no 0x48 which I've gathered from my error-seeking that I should be seeing.

The error I get is: OSError; [Errno 5] Input/output error on some part of the LM75B-stuff. Which makes sense if it isn't detected.

I've done as good a job soldering the header as I can with my current iron. Anything more I should look into before judging this papirus broken? Initially I wasn't soldering the headers, then I didn't get 0x6f either.

shawaj, 2018-03-24:
Guessing this is a PaPiRus Zero yes?

Can you send some pictures of your setup?

lawik, 2018-03-24:
Yup, a zero, 2.7 inch.

Appreciate the quick response but I'm going travelling for a bit so I'll have to get back to you in a few weeks if that's alright.

shawaj, 2018-03-24:
It can't be a zero if it's 2.7 inch.

Can you send some pictures of your setup?

lawik, 2018-03-25:
From pimoroni the order said: PaPiRus Zero – ePaper / eInk Screen pHAT for Pi Zero
The box says: 2.7"

[image]
[image]
[image]

tvoverbeek, 2018-03-25:
You can sure use a PaPiRus-HAT on a Pi-Zero.
Looking at the pictures (Many streaks on the display and excessive solder for the 40 pin GPIO connector on the Zero) it seems a matter of bad contacts between the HAT and the Pi-Zero.
Double check the I2C, SPI and other HAT contacts.
See [https://pinout.xyz/pinout/papirus_hat](url) for all pins used by the PaPiRus HAT

shawaj, 2018-03-27:
It looks like also maybe the soldering is not too good on the header. Do you have another Pi with a pre-soldered header you could test on like a Pi3 or Pi2 or something?

lawik, 2018-03-28:
Yeah, I have a clumsy iron and am pretty new at soldering. I recently got a spare Pi 3 that I could try it with I guess. I'll try with that and I should find out if the Papirus works. In that case I should just have to redo the soldering for the header right?

I can't try this for a few weeks so I'll close the issue for now and open it back up if it doesn't work out.

lawik, 2018-04-15:
Hi, I've been trying it now with a full-size Pi with a pre-soldered header and I'm seeing the same behavior. 

Install of everything was fine. Made sure I2C and SPI are on. Ran papirustest, got a temperature reading. And then it choked on "animation using partial update" when trying to work with the LM75B-stuff. Running i2cdetect -y 1 just gives the 6f and on a second run of papirus-test it just crashes out straight away to an input/output error.

lawik, 2018-04-18:
@shawaj I've tried your suggestion so I figure I'll give a ping before getting in touch with Pimoroni about a possible return.