PiSupply/PaPiRus #185: C language library?

uchuugaka, 2018-04-27:
Where can we find C or other compiled language library for the PaPiRus products? I need to do things not in Python sometimes...

tvoverbeek, 2018-05-05:
The low level driver is just a file interface (FUSE based) writing/reading to various files under /dev/epd.
See https://github.com/PiSupply/PaPiRus/blob/master/bin/papirus-setup for how this is installed.
The driver itself can be found in https://github.com/repaper/gratis.

Have a look at some of the python scripts to see how this interface is used.
For example clearing the display is done by writing the character 'C' to /dev/epd/command
Using it from C/C++ should be no problem.

uchuugaka, 2018-05-05:
Awesome!
Thanks so much for the insights. 
I will dig into these.