PiSupply/PaPiRus #186: smbus install issues

MartenKL, 2018-05-03:
When running;
"sudo python3 setup.py install"
I encounter an error;
"ModuleNotFoundError: No module named 'smbus'"

I have tried installing smbus and smbus2 with pip, pip3 pip3.6 no combination worked.
running;
"sudo apt-get install python3-smbus"
works and installs smbus but only for Python 3.5.3, not Python 3.6.5
Is there a simple way to copy the library from 3.5.3 to 3.6.5?

tvoverbeek, 2018-05-04:
The raspbian python3-smbus installs in /usr/lib/python3/dist-packages
```
pi@raspberrypi:~ $ dpkg-query -L python3-smbus
/.
/usr
/usr/lib
/usr/lib/python3
/usr/lib/python3/dist-packages
/usr/lib/python3/dist-packages/smbus.cpython-35m-arm-linux-gnueabihf.so
/usr/share
/usr/share/doc
/usr/share/doc/python3-smbus
/usr/share/doc/python3-smbus/changelog.Debian.gz
/usr/share/doc/python3-smbus/changelog.gz
/usr/share/doc/python3-smbus/copyright
pi@raspberrypi:~ $ 
```
In your Python 3.6 do:
```
import sys
print(sys.path)
```
Check if `/usr/lib/python3/dist-packages` is included in the path.
If so `import smbus` should work in python 3.6 as far as I know.

MartenKL, 2018-05-04:
It is not in my path but when I add it I get the same result:
pi@ZeroPaper:~ $ python3
Python 3.6.5 (default, May  3 2018, 21:34:56)
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> print(sys.path)
['', '/usr/local/lib/python36.zip', '/usr/local/lib/python3.6', '/usr/local/lib/python3.6/lib-dynload', '/usr/local/lib/python3.6/site-packages']
>>> sys.path.insert(0,"/usr/lib/python3/dist-packages")
>>> print(sys.path)
['/usr/lib/python3/dist-packages', '', '/usr/local/lib/python36.zip', '/usr/local/lib/python3.6', '/usr/local/lib/python3.6/lib-dynload', '/usr/local/lib/python3.6/site-packages']
>>> import smbus
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'smbus'
>>> exit()
pi@ZeroPaper:~ $ dpkg-query -L python3-smbus
/.
/usr
/usr/lib
/usr/lib/python3
/usr/lib/python3/dist-packages
/usr/lib/python3/dist-packages/smbus.cpython-35m-arm-linux-gnueabihf.so
/usr/share
/usr/share/doc
/usr/share/doc/python3-smbus
/usr/share/doc/python3-smbus/changelog.Debian.gz
/usr/share/doc/python3-smbus/changelog.gz
/usr/share/doc/python3-smbus/copyright

tvoverbeek, 2018-05-04:
smbus2 on PyP1 is supported for python 3.6.
You will need to change all `import smbus` statements to `import smbus2`.

Alternatively you can look at my py-smbusf in https://github.com/PiSupply/PaPiRus/tree/master/RTC-Hat-Examples/py-smbusf
In setup.py change the module name from smbusf to smbus, then you do not have to change the scripts.