PiSupply/PaPiRus #190: Papirus in virtual environment

mtaylor-create, 2018-06-09:
Hi all,
  Little new to all of this, so I apologize if this has already been covered or has an easy solution.  I'd like to get the python library working in a virtual environment that already has some other stuff in it.  It seems like the setup scripts install the library properly outside of the environment, and I can import papirus without any errors.  Inside of the virtual python environment, though, I get:

ImportError: No module named 'papirus'

  Normally I'd try to do a pip install from inside the environment but it sounds like that may not be an option yet, from what I've read so far.  Any suggestions for getting this to work inside the environment?  Could it be as simple as copying some directories?  I'm using the Papirus Zero with a Pi Zero W, and the virtual env I'm using is with Python 3.5.3.  Thanks!

tvoverbeek, 2018-06-09:
First, I am not an expert on Python virtual environments.

As far as I can tell if you run the setup.py script (https://github.com/PiSupply/PaPiRus/blob/master/setup.py) inside your virtual environment) `python setup.py install` it should install the papirus module in the right place.
You probably have to remove the scripts and data_files lines from setup.py (do not know how they are treated in a virtual environment).

The epd-fuse service still has to be running outside your virtual environment.

21isenough, 2019-08-31:
You can resolve this problem like this:

Clone the Github repo to your local hard drive with `git clone --depth=1 https://github.com/PiSupply/PaPiRus.git`. Then `cd PaPiRus` into the folder just created. 

With your virtual python environment activated you now run `which python`. This will show you your virtualenv python instance e.g. /home/user/env/bin/python.

Then you have to execute your "setup.py install" command by specifying the binary discovered above like this:  `/home/user/env/bin/python setup.py install` 

This will install the module papirus into your virtual environment.