PiSupply/PaPiRus #192: ${LDFLAGS} Placement requires moving for fuse driver to compile

peterberbec, 2018-06-20:
I'm using cc (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.9) 5.4.0 20160609.

In `gratis/PlatformWithOS/driver-common/Makefile`, the `${LDFLAGS}` needs to be moved to the end of the `${CC}` command:

```# build the fuse driver
CLEAN_FILES += epd-fuse
epd_fuse: ${FUSE_OBJECTS}
        ${CC} ${CFLAGS} ${LDFLAGS} -o "$@" ${FUSE_OBJECTS}

# build simple GPIO test program
CLEAN_FILES += gpio_test
gpio_test: ${GPIO_OBJECTS}
        ${CC} ${CFLAGS} ${LDFLAGS} -o "$@" ${GPIO_OBJECTS}

# build EPD test program
CLEAN_FILES += epd_test
epd_test: ${TEST_OBJECTS}
        ${CC} ${CFLAGS} ${LDFLAGS} -o "$@" ${TEST_OBJECTS}
```

turns into

```# build the fuse driver
CLEAN_FILES += epd-fuse
epd_fuse: ${FUSE_OBJECTS}
        ${CC} ${CFLAGS} -o "$@" ${FUSE_OBJECTS} ${LDFLAGS}

# build simple GPIO test program
CLEAN_FILES += gpio_test
gpio_test: ${GPIO_OBJECTS}
        ${CC} ${CFLAGS} -o "$@" ${GPIO_OBJECTS} ${LDFLAGS}

# build EPD test program
CLEAN_FILES += epd_test
epd_test: ${TEST_OBJECTS}
        ${CC} ${CFLAGS} -o "$@" ${TEST_OBJECTS} ${LDFLAGS}
```

This seems to be a common issue if the linker options are placed before the files being compiled.

tvoverbeek, 2018-06-20:
No need to move LD_FLAGS when using gcc on Raspberry Pi.
Is this a cross compiler for ARM?

peterberbec, 2018-06-20:
I'm using an ["alternative" Pi.](www.banana-pi.org/bpi-zero.html)

peterberbec, 2018-06-20:
And... it looks like I'll have to rewrite parts of the gratis library :)

Let me get that working and i'll come back. Sorry to bother you prematurely.

closed 2018-06-20 by peterberbec