PiSupply/PaPiRus #194: Fix and update README.md

tahouse, 2018-08-02:
I appreciate the attempt at providing API examples, but the current [README.md](https://github.com/PiSupply/PaPiRus/blob/master/README.md) examples do not work. Running any of the examples, a new user will be greeted immediately with code that simply doesn't work.

 Let's look at the Python API example
```from papirus import Papirus

# The epaper screen object.
# Optional rotation argument: rot = 0, 90, 180 or 270
screen = Papirus([rotation = rot])

# Write a bitmap to the epaper screen
screen.display('./path/to/bmp/image')

# Perform a full update to the screen (slower)
screen.update()

# Update only the changed pixels (faster)
screen.partial_update()

# Update only the changed pixels with user defined update duration
screen.fast_update()

# Disable automatic use of LM75B temperature sensor
screen.use_lm75b = False``` 

In the line `screen = Papirus([rotation = rot])`, [rotation = rot] isn't valid Python syntax.

With your current module specification `screen.display('./path/to/bmp/image')` also doesn't work. It appears from the module code that it's actually expecting an image bytestream, not a string path to an image. There is also no mention of resizing an image to the correct size/resolution for the screen set in the setup/driver configuration.

vitasam, 2018-08-02:
Hi,
> In the line screen = Papirus([rotation = rot]), [rotation = rot] isn't valid Python syntax.

This syntax works:
```
from papirus import PapirusText
text = PapirusText(rotation = 90)
text.write("Hello World!")
```

But I tend to agree that README.md could be refactored

ChristopherRush, 2018-08-02:
@tahouse I will re-word the documentation and close this issue.

ChristopherRush, 2018-08-06:
dd9a12263cfa252a55e87556f3f8d6b8160f1424

tvoverbeek, 2018-08-06:
The `screen.display('./path/to/bmp/image')` issue has not been addressed.
Showing an image from file is in the PapirusImage class.
The `screen.display` expects an PIL image structure.