PiSupply/PaPiRus #195: epd_fuse CPU usage

AndyMo905, 2018-08-10:
Hi there,

Running a really simple python script (display elapsed seconds) on a PiZero with the PaPiRus Zero.  I have it updating the screen using partial updates, and have it looped so that every 5 minutes or so it does a full refresh.

The CPU usage of epd_fuse is pretty high, but after a couple of days things get really bogged down and the Pi CPU is at 100%.  At that point updates and mouse control slow right down.  At day 3-4 I get maybe 3-5 screen updates a day because the PiZero is slammed.

Is there a clean-up routine that should be done to help ?

Thanks !

Andy

tvoverbeek, 2018-08-10:
Are you sending partial updates continuously?
Since you want to display elapsed seconds, you can sleep for about a second between updates.
If you do not sleep epd-fuse will be running behind since executing the previous update request is not finished before the next one comes in. This might explain the high CPU usage and running out of memory (slow Pi).
For example look at the code in papirus-clock (https://github.com/PiSupply/PaPiRus/blob/master/bin/papirus-clock) where a partial update is only sent when the second changes.

AndyMo905, 2018-08-12:
tvoverbeek - that example code was perfect !  In addition to getting things working more smoothly, it helped me avoid additional apis like positional text, which i think were also slowing things down.

Thank you very much for your help.  

Andy