PiSupply/PaPiRus #196: Sorry: TabError: inconsistent use of tabs and spaces in indentation (textpos.py, line 20)

eat-sleep-code, 2018-08-18:
I attempted to execute `curl -sSL https://pisupp.ly/papiruscode | sudo bash` and chose the Python 2 and Python 3 option.

I noted the following error being thrown:

> Sorry: TabError: inconsistent use of tabs and spaces in indentation (textpos.py, line 20)

tvoverbeek, 2018-08-18:
@eat-sleep-code The indentation mix of spaces and tabs in textpos.py was fixed in github on October 7, 2017. See commit https://github.com/PiSupply/PaPiRus/commit/36a711c752b372c70e8783767dd88db374f1e617
So I do not understand how you get this error now.
Previous install still hanging around, so pip3 did not install the current version?

eat-sleep-code, 2018-08-22:
Hmm, I reinstalled using the script, maybe I will try to manually do it.

eat-sleep-code, 2021-04-18:
This is how to resolve:

* Run:
```
sudo rm -Rf ~/PaPiRus
sudo rm -Rf /usr/local/lib/python3.7/dist-packages/papirus
curl -sSL https://pisupp.ly/papiruscode | sudo bash
```
* Select Python 3 option
* Select Screen Option
* Reboot when prompted

closed 2021-04-18 by eat-sleep-code