PiSupply/PaPiRus #197: Switch 4 pressed in PaPiRus Zero 2.0''

DrumSergio, 2018-09-15:
Hello!

I am just running a simple program that outputs which switch is pressed in the log.
And the problem is that, whenever I turn the Pi Zero up, SW4 is pressed a couple of times. Don't know why.

I've tried with another Pi and also the same happens. Then everything runs smoothly. But at first, it's like crazy. Don't know if you've suffered the same behaviour.

Thank you!

shawaj, 2018-09-15:
This sounds like a floating button type scenario or some kind of denounce issue. Or possibly a sticky button

Can you post the code you're using?

tvoverbeek, 2018-09-16:
SW4 is connected to GPIO19 (pin 35). This is also used as the LRCLOCK (Left-Right clock) for I2S sound peripherals. @DrumSergio Are you initializing some I2S peripheral at boot? That would explain the multiple SW4 presses.

DrumSergio, 2018-09-21:
`
``#!/usr/bin/python3

import RPi.GPIO as GPIO
import requests
from papirus import PapirusImage
from papirus import PapirusTextPos
import json
import time
import os
from datetime import datetime
import sys

def click(pressed_channel):
    for channel in channels:
        GPIO.remove_event_detect(channel)
    print("Switch " + str(switches.index(pressed_channel) + 1) + " pressed")
    sys.stdout.flush()

    ## Get data
    duration="0"
    if pressed_channel==21: duration="1"
    elif pressed_channel==16: duration="2"
    elif pressed_channel==20: duration="3"
    elif pressed_channel==19: duration="4"
    elif pressed_channel==26: os.system('sudo reboot'); return

    ## More lines here... Just putted them out for the example

    text = PapirusTextPos(False, rotation = rot)
    text.AddText("Duration: " + duration, 10, 10, size=18, Id="Top")
    text.WriteAll()

    finally: # We guarantee this chunk of code to run even if an exceptions rises up
        for channel in channels:
            GPIO.add_event_detect(channel, GPIO.RISING, callback=click, bouncetime=200)

    return

def main():

    now = datetime.now()
    print("Python3 script started at " + now.strftime("%Y-%m-%d %H:%M:%S"))
    sys.stdout.flush()

    global rot
    global switches
    global wd
    global channels

    rot = 0
    switches = [21, 16, 20, 19, 26] # The list of the switches ordered by Switch Number
    wd = os.path.dirname(os.path.realpath(__file__))

    image = PapirusImage(rotation = rot)
    image.write(wd + '/logo.png')

    GPIO.setmode(GPIO.BCM)
    channels=[16, 19, 20, 21, 26] # SW2 - BCM16. The rest: BCM19 - SW4, BCM20 - SW3, BCM21 - SW1, BCM26 - SW5
    GPIO.setup(channels, GPIO.IN)

    for channel in channels:
        GPIO.add_event_detect(channel, GPIO.RISING, callback=click, bouncetime=200)

    while True:
        time.sleep(60)

if __name__ == "__main__":
    main()`

DrumSergio, 2018-09-21:
```
#!/usr/bin/python3

import RPi.GPIO as GPIO
import requests
from papirus import PapirusImage
from papirus import PapirusTextPos
import json
import time
import os
from datetime import datetime
import sys

def click(pressed_channel):
    for channel in channels:
        GPIO.remove_event_detect(channel)
    print("Switch " + str(switches.index(pressed_channel) + 1) + " pressed")
    sys.stdout.flush()

    ## Get data
    duration="0"
    if pressed_channel==21: duration="1"
    elif pressed_channel==16: duration="2"
    elif pressed_channel==20: duration="3"
    elif pressed_channel==19: duration="4"
    elif pressed_channel==26: os.system('sudo reboot'); return

    ## More lines here... Just putted them out for the example

    text = PapirusTextPos(False, rotation = rot)
    text.AddText("Duration: " + duration, 10, 10, size=18, Id="Top")
    text.WriteAll()

    finally: # We guarantee this chunk of code to run even if an exceptions rises up
        for channel in channels:
            GPIO.add_event_detect(channel, GPIO.RISING, callback=click, bouncetime=200)

    return

def main():

    now = datetime.now()
    print("Python3 script started at " + now.strftime("%Y-%m-%d %H:%M:%S"))
    sys.stdout.flush()

    global rot
    global switches
    global wd
    global channels

    rot = 0
    switches = [21, 16, 20, 19, 26] # The list of the switches ordered by Switch Number
    wd = os.path.dirname(os.path.realpath(__file__))

    image = PapirusImage(rotation = rot)
    image.write(wd + '/logo.png')

    GPIO.setmode(GPIO.BCM)
    channels=[16, 19, 20, 21, 26] # SW2 - BCM16. The rest: BCM19 - SW4, BCM20 - SW3, BCM21 - SW1, BCM26 - SW5
    GPIO.setup(channels, GPIO.IN)

    for channel in channels:
        GPIO.add_event_detect(channel, GPIO.RISING, callback=click, bouncetime=200)

    while True:
        time.sleep(60)

if __name__ == "__main__":
    main()
```

tvoverbeek, 2018-09-21:
@DrumSergio Note that the PaPiRus Zero has hardware pull-up resistors. So the non-pressed input is high. Pressing the switch connects the input to earth.
See e.g https://github.com/PiSupply/PaPiRus/blob/master/hardware/PaPiRus%20Zero/button-schematic.jpg
Your spurious SW4 might jsut be the consequence of the PaPiRus zero power up (SW4 GPIO input going high).
Change the GPIO.RISING to GPIO.FALLING.
Hopefully that solves the problem.

DrumSergio, 2018-09-21:
Thank you very much! That just solved the issue. Now I understand more about the pull-up resistor thanks to you!

closed 2018-09-21 by DrumSergio