PiSupply/PaPiRus #198: Clear the screen before shutdown

DrumSergio, 2018-09-22:
I was trying to clear the PaPiRus screen before shutdown/halt/reboot. Sometimes it's hard to know if the Pi is switched on or switched off! So I think this could be a great feature for everybody... If it can be solved.

I am using Raspbian Lite and a PaPiRus Zero in a Pi Zero W.

At first, I tried installing some scripts under `/lib/systemd/system-shutdown/` without success. I tried first copying the `papirus-clear` script located at `/usr/local/bin/papirus-clear`. Then I created a bash script just pointing to it. 

As I didn't have success, I wanted to experiment with Systemd. This is the last iteration after installing and enabling it (when you do a `service X status` all seems ok):

```
[Unit]
Description=Run papirus-clear at shutdown

[Service]
Type=oneshot
RemainAfterExit=yes
ExecStart=/bin/true
ExecStop=/usr/bin/python3 /usr/local/bin/papirus-clear

[Install]
WantedBy=multi-user.target
``` 

I tried also adding the lines:
```
DefaultDependencies=no
Before=shutdown.target reboot.target
```
under the [Unit] section but it doesn't work.

I would like to know how could I do it. Maybe the error is in the timing, because the Pi shuts down very fast, and the `papirus-clear` script takes a lot of time to execute...

Anyway, just to be clear: I've tested at the CLI `/usr/bin/python3 /usr/local/bin/papirus-clear` and it works, but not anywhere else.

I don't know. Thank you very much for the help!

tvoverbeek, 2018-09-22:
To be able to run papirus-clear it needs the epd-fuse service to be active. It is probably already terminated before your service runs at shutdown. Try to add a dependency on epd-fuse and an after epd-fuse clause, so at shutdown your service is run before epd-fuse is terminated.

DrumSergio, 2018-09-22:
Thank you very much!

I am searching for "epd-fuse" in the repository because in my instance there's no service for epd-fuse, but in the code I can see sometimes `systemctl start epd-fuse`... It's strange. Do I need to create a service file for epd-fuse? How it is started if not?

I also changed the file to include `After=network.target` without success. 

Is there anything like `After=getprocessname(epd_fuse)`? I've tried this binding it to /dev/epd without success: https://unix.stackexchange.com/questions/186899/how-to-wait-for-a-dev-video0-before-starting-a-service

Thank you!

tvoverbeek, 2018-09-23:
@DrumSergio You were almost there.

The epd-fuse service is installed as part of the PaPiRus install.
The service creates a /dev/epd device which is a 'file system in user space (fuse)'.
By writing to and reading from it you command the display. The nitty gritty of all the spio transactions
needed is handled by the fuse file system.
See the papirus-install script https://github.com/PiSupply/PaPiRus/blob/master/bin/papirus-setup.
The actual service code is in the external repository https://github.com/repaper/gratis.git which is called from papirus-install.
The epd-fuse.service file is in /usr/lib/systemd/system.

I created a papirus-clear.service file and  also put it in /usr/lib/systemd/system.
Its contents:
```
[Unit]
Description=Run papirus-clear at shutdown
Wants=epd-fuse.service
After=epd-fuse.service

[Service]
Type=oneshot
RemainAfterExit=yes
ExecStart=/bin/true
ExecStop=/usr/local/bin/papirus-clear

[Install]
WantedBy=multi-user.target
```
Almost your file, only added dependency on epd-fuse so at shutdown/reboot the ExecStop gets called before the epd-fuse service is terminated (reverse order at shutdown/reboot).
Do not forget to call `sudo systemctl daemon-reload` after adding the papirus-clear.service file.

Tested it by running papirus-clock and then calling shutdown and reboot.
In both cases the screen clears.