PiSupply/PaPiRus #201: Writing variables to papirus

plagergren, 2018-12-08:
I am writing some lua code that has the following where float is a floating point number:
floater = tostring(float)
os.execute(sudo paprius-textfill 'floater')

and this works fine except I want the value of floater to display not the word "floater"
I have tried a lot of different ways to do what I want including grep and all kinds of folderol. Nothing works so far.  I get all kinds of different errors depending on which code du jour I am using.
How do I assign a value and print it to the papirus screen using textfill? 
Actually, what I really want to do is to use screen.partial_update to refresh only part of the screen that has changed.

Thanks!

plagergren, 2018-12-14:
> @plagergren
> floater is already a string.
> Try with os.execute('papirus-textfill '..floater)
> .. is the lua string concatenation operator.
> For papirus sudo is not needed unless you have an old version of Raspbian.
> Here a sample lua test program which works for me:
> for val=1.0,2.0,0.1 do
>    cmd='papirus-textfill '..tostring(val)..' 180'
>    os.execute(cmd)
>    os.execute('sleep 1')
> end
>
> The 180 is for rotating the image on the display.
> For partial updates you either need some python glue or implement it directly in lua.
> See the papirus-clock example for how to use papirus.partial_update().

HI Ton, 

Got my code working except for one thing - I can't get partial_update to work. When I started using it I got error messages of several kinds but all clustered around 'command not found' which isn't a surprise. However, this lua code does not cause an error: 

floater = string.format("$07.1f"float) 
cmd = "papirus-textfill"..floater 
os.execute(cmd) -- this works splendidly 
float = float + 1.0 
floater = string.format("$07.1f"float) 
cmd = "partial_update() "..floater -- this does not generate a "not found" error 
os.execute(cmd) -- this has no effect on the display 

After looking at the papirus-clock code I am curious if the reason I'm not seeing any activity from partial_update is because I am not treating the displayed numbers as an image. Does partial_update work on text or only on images? 

Best regards, 

Peter

tvoverbeek, 2018-12-14:
@plagergren The papirus example programs use the papirus python module as interface between the low level driver (the epd_fuse service) and the program.
In principle all commands are sent to files in /dev/epd. You would have to implement the papirus module in lua to get a pure lua implementation.
If you can call a python script from lua you could write a small one which does the partial_update
and call that from your lua program.
You will have to dive into the low level details to get this working.

plagergren, 2018-12-14:
Thanks! I'll dive into it and see where it takes me.
Pete

https://github.com/PiSupply/PaPiRus/issues/201#issuecomment-447432523

plagergren, 2018-12-27:
Hi Ton, 

I've been playing around with the papirus and one of the things I see is that the partial update examples are all images. I am displaying ascii numbers using the papirus-texfill 
which works just fine from lua. However, it seems that I can't do a partial update because the displayed figures are not images. Do I need to convert the text to an image before I can do a partial update? 

Thanks in advace 

Peter

tvoverbeek, 2018-12-27:
@plagergren Yes, all text displayed is first made as an image using the Python Image Library and then copied to the PaPiRus.
I just updated papirus-textfill on github with an option to use partial-update.
Here is the new help:
```
usage: papirus-textfill [-h] [--rotation ROTATION] [--partialupdate]
                        displaytext

positional arguments:
  displaytext           Text to display (max 40 char)

optional arguments:
  -h, --help            show this help message and exit
  --rotation ROTATION, -r ROTATION
                        Rotation one of 0, 90, 180,270
  --partialupdate, -p   Use partial instead of full update
```
Just replace your copy of papirus-textfill with the current one on github and use the '-p' option.

PS Your name sounds scandinavian. Swedish? If so Lycka till

plagergren, 2018-12-27:
Hej Tom,

Tak so mykjet! 

I was born in Pitea and emigrated to the states when I was 5 so tragically I’ve lost my Swedish language ability. Not that a 5 year old has much.
Hej do!

https://github.com/PiSupply/PaPiRus/issues/201#issuecomment-450170722