PiSupply/PaPiRus #206: SW3 is not working

starkshaw, 2019-03-27:
Hi,

The SW3 button on my board is not working. The rest buttons are fine, but not SW3. If I press this button, the screen will freeze and I have to power off the board and plug it back in again to use the screen again.

Note that, not reboot, I have to cut the power supply to make it work again.

Here are the error messages when I press SW3:
```
$ papirus-test
Assuming this is a PaPiRus HAT board

---- Detect temperature sensor and RTC (RTC for HAT only) ----
NXP LM75BD detected at: 0x48
MCP7940N detected at: 0x6f

---- Screen Information ----
Vendor: Pi Supply
Product: PaPiRus ePaper HAT v1.91
Serial: 342d3161-3531-2d35-3166-32303332357b
Date/Time from RTC: Monday 01 January 2001 - 00:00:00
Temperature from sensor: 38 degrees Celsius
Panel info: EPD 2.7 264 x 176 version=4 COG=2 FILM=231

Screen status OK
Displaying an image using full update
Displaying an animation using partial update
Displaying information about the board and screen

Board layout SW4--SW3--SW2--SW1
Board pins    36   37   38   40
BCM GPIO      16   26   20   21
Press SW4 for board info, SW3 for animation, SW2 for single image, SW1 to exit
SW1 Pressed - Exiting
Traceback (most recent call last):
  File "/usr/local/bin/papirus-test", line 285, in <module>
    main(sys.argv[1:])
  File "/usr/local/bin/papirus-test", line 112, in main
    papirus.clear()
  File "/usr/local/lib/python3.5/dist-packages/papirus/epd.py", line 206, in clear
    self._command('C')
  File "/usr/local/lib/python3.5/dist-packages/papirus/epd.py", line 211, in _command
    f.write(b(repr(self._lm75b.getTempC())))
  File "/usr/local/lib/python3.5/dist-packages/papirus/lm75b.py", line 39, in getTempC
    raw = self._bus.read_word_data(self._address, LM75B_TEMP_REGISTER) & 0xFFFF
OSError: [Errno 121] Remote I/O error
```

Interestingly, it provided a false message of SW1 pressed.

These are the error messages if I run `papirus-buttons` and press SW3:
```
$ papirus-buttons
Traceback (most recent call last):
  File "/usr/local/bin/papirus-buttons", line 146, in <module>
    main(sys.argv[1:])
  File "/usr/local/bin/papirus-buttons", line 87, in main
    write_text(papirus, "Exiting ...", SIZE)
  File "/usr/local/bin/papirus-buttons", line 142, in write_text
    papirus.partial_update()
  File "/usr/local/lib/python3.5/dist-packages/papirus/epd.py", line 200, in partial_update
    self._command('P')
  File "/usr/local/lib/python3.5/dist-packages/papirus/epd.py", line 211, in _command
    f.write(b(repr(self._lm75b.getTempC())))
  File "/usr/local/lib/python3.5/dist-packages/papirus/lm75b.py", line 39, in getTempC
    raw = self._bus.read_word_data(self._address, LM75B_TEMP_REGISTER) & 0xFFFF
OSError: [Errno 121] Remote I/O error
```

If I test other buttons on `papirus-test`, I get this:
```
$ papirus-test
Assuming this is a PaPiRus HAT board

---- Detect temperature sensor and RTC (RTC for HAT only) ----
NXP LM75BD detected at: 0x48
MCP7940N detected at: 0x6f

---- Screen Information ----
Vendor: Pi Supply
Product: PaPiRus ePaper HAT v1.91
Serial: 342d3161-3531-2d35-3166-32303332357b
Date/Time from RTC: Monday 01 January 2001 - 00:00:00
Temperature from sensor: 38 degrees Celsius
Panel info: EPD 2.7 264 x 176 version=4 COG=2 FILM=231

Screen status OK
Displaying an image using full update
Displaying an animation using partial update
Displaying information about the board and screen

Board layout SW4--SW3--SW2--SW1
Board pins    36   37   38   40
BCM GPIO      16   26   20   21
Press SW4 for board info, SW3 for animation, SW2 for single image, SW1 to exit
SW4 Pressed
SW2 Pressed
SW1 Pressed - Exiting
```

If I don't run the testing app but pressing the SW3 button, the screen will freeze as well.

So, I wonder if there are some issues in the driver that when I press the button some exceptions have been thrown.

Also, I didn't attach the pogo pin because I am using the Raspberry Pi 3B+. The pin does not fit the location on this board.

Thanks!

tvoverbeek, 2019-03-27:
There is obviously something wrong with SW3. papirus-buttons exits when SW1 and SW2 are pressed simultaneously. So it seems when you press SW3 it is detected as a press of both SW1 and SW2 (and probably also of SW3 and SW4).
Also pressing SW3 causes the I2C bus to hang. Probably it causes some form of short.

shawaj, 2019-03-29:
As this is a PaPiRus HAT - I guess you soldered the buttons on yourself?

Could you check the solder joints are all ok? Perhaps send a picture so we can take a look.