PiSupply/PaPiRus #207: Dependencies for CentOS

osowp, 2019-06-07:
Would it be possible to get some help figuring out the dependencies needed for CentOS.  CentOS uses Yum not Apt-Get.

tvoverbeek, 2019-06-08:
I am assuming you are using CentOS on Raspberry Pi.
The display uses SPI. So first thing to check is how SPI support (/dev/spidev0.0 and /dev/spidev0.1) is enabled. I suppose using a device overlay parameter (dtparam) in /boot/config.txt.
The driver (from https://github.com/repaper/gratis) uses libfuse (file system in user space).
I believe you can install fuse support with 'yum install fuse fuse-dev'.
Then check the build procedure in https://github.com/repaper/gratis for RPi. If you are lucky you can used the Makefiles unchanged.
Hope that gets you started.

osowp, 2019-06-09:
Thank you TVoverbeek  but I was not even up to that step when I wrote the above.   I have since found the following conversions of Apt-get to yum install.   I am still missing the Font package at this point.

    "development tools"  installs several packages like: git, GCC, make,  
    https://www.cyberciti.biz/faq/centos-rhel-7-redhat-linux-install-gcc-compiler-development-tools/

    yum install bc
    yum install python-pillow
    yum install i2c-tools-python
    yum install python-dateutil
    yum install fuse-devel

Not confirmed if these are needed:
    yum install fuse-libs
    yum install fuse

After installing the above I think I should be able to at least run the papirus-clear.   Which currently gives me this error:   smbus.SMBus(busnum)  no such file or directory.   
Which is probably related to TVOver's answer above

tvoverbeek, 2019-06-09:
I suppose 'yum install fuse-devel' will pull in fuse-libs and fuse as dependencies.
Regarding the 'smbus.SMBus(busnum) no such file or directory.' error.
The smbus python package is for i2c support.
As far as I know this is included in i2c-tools-python for Centos.
papirus needs i2c to communicate with the lm75b temperature sensor on the papirus board.
When you run python manually can you import the smbus package without errors?
I suppose you need to enable i2c bus access somehow.

osowp, 2019-06-09:
Just got the i2c enabled with  https://lists.centos.org/pipermail/arm-dev/2017-January/002628.html

> We need to activate it through config.txt :
> 
> 
> echo -e "dtparam=i2c1=on\ndtparam=i2c_arm=on" >> /boot/config.txt
> echo -e "i2c-bcm2708\ni2c-dev" >> /etc/modules-load.d/i2c.conf
> yum install -y i2c-tools
> systemctl reboot
> 
> Once rebooted :
> 
> [root at rpi2 ~]# i2cdetect -y 1
>      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
> 00:          -- -- -- -- -- -- -- -- -- -- -- -- --
> 10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 70: -- -- -- -- -- -- -- --

osowp, 2019-06-09:
Next error:  "no such file or directory:  '/dev/epd/version'

tvoverbeek, 2019-06-09:
Then the epd driver from https://github.com/repaper/gratis is not installed or not working.
Did you manage to compile and build it?
The resulting epd-fuse is a filesystem in userspace which creates the /dev/epd directory.
It is started as a systemctl service at boot.
See the 'epd-fuse.service' file in https://github.com/repaper/gratis/blob/master/PlatformWithOS/driver-common/epd-fuse.service
It is installed with the Make install
Do not forget to enable spi in /boot/config.txt: `dtparam=spi=on`

osowp, 2019-06-09:
It looks like the gratis built correctly, (just did it again to be sure)

but the "sudo systemctl start epd-fuse"  failed

tvoverbeek, 2019-06-09:
Does this mean you got papirus-clear working?

osowp, 2019-06-09:
Partly,   Papirus-clear is "trying" to clear the screen.   but its leaving the out of sync lines.   which I thought i had seen as a known issue before but can't find it now

tvoverbeek, 2019-06-09:
That probably means you have not set the display type/size correctly in /etc/default/epd-fuse.
Run the papirus-config script. For me it is in /usr/local/bin

osowp, 2019-06-09:
I had modified -clear to hardcode the EPD size to 2.7

But now that I run -config I get an invalid syntax error on line 4
    If [[ $EUID -ne 0 ]]; then

tvoverbeek, 2019-06-09:
it assumes it is run as a bash script. Is the environment variable EUID (effective user id) set?
Check in a bash shell `echo $EUID`.
It is a check that the scritp is running as root.

osowp, 2019-06-09:
sorry  my brain must be fried.    stupid mistake.       I was trying to run it as a python program

tvoverbeek, 2019-06-09:
Regarding the fonts. FreeMono.ttf etc are in
https://centos.pkgs.org/7/slce-x86_64/freefont-20120503-1.el7.noarch.rpm.html

osowp, 2019-06-09:
for those that follow me on centos there is one change and one extra step to the centos..freefont 

1)  for the slce.repo:   replace "$basearch" with "x84_64" 
Create the repository config file /etc/yum.repos.d/slce.repo:
[slce]
name=Scientific Linux Cyrillic Edition
baseurl=http://downloads.naulinux.ru/pub/SLCE/7x/**$basearch**/CyrEd/RPMS/
enabled=0
gpgcheck=1
gpgkey=http://downloads.naulinux.ru/pub/SLCE/RPM-GPG-KEY-linux-ink
Install freefont rpm package:
# yum --enablerepo=slce install freefont

2)move the installed fonts to where they are supposed to be: 
    cd /usr/share/fonts
    mkdir truetype
    mv freefont truetype