PiSupply/PaPiRus #212: Using PaPiRus HAT with PiJuice HAT on Pi4, no eink output?

andrewsmyname, 2019-12-31:
I was told by one of your colleagues to post here, and he did his best to help out (Ryan Walmsley). He deserves a pint!

He let me know that the EEPROM of both HATs collied, so I changed the PiJuice EEPROM to 52, which seemed to improve matters as the `papyrus-test` command now correctly identifies the Pi as not a Pi Zero. But nothing shows up on the e-ink display (I'm using the full size HAT with the 2.0 inch display, and have checked that it is seated correctly).

Here's some outputs to help:

I was told by one of your colleagues to post here, and he did his best to help out (Ryan Walmsley). He deserves a pint!

He let me know that the EEPROM of both HATs collied, so I changed the PiJuice EEPROM to 52, which seemed to improve matters as the `papyrus-test` command now correctly identifies the Pi as not a Pi Zero. But nothing shows up on the e-ink display (I'm using the full size HAT with the 2.0 inch display, and have checked that it is seated correctly).

Here's some outputs to help:

```bash
pi@calipi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- 6f
70: -- -- -- -- -- -- -- --
```

As mentioned in the PiJuice docs, when changing the EEPROM from 50 to 52, `i2c-rtc,ds1339` will have to be added to `/boot/config.txt`. I also note that `i2c-bcm2708` is an entry - removing it stops me from being able to communicate with the PiJuice.
`dtoverlay=i2c-bcm2708,i2c-rtc,ds1339`

The PaPiRus also has an RTC that I don't need to use, but to enable that I have to change that entry to:
`dtoverlay=i2c-rtc,mcp7940x=1`

This allows that RTC to actually function and be read with `sudo hwclock -r`, but of course without `i2c-bcm2708` I can't communicate with the PiJuice (and using `dtoverlay=i2c-bcm2708,i2c-rtc,mcp7940x=1` results in the RTC not working).

RTC aside, the main issue is that the e-ink display doesn't show anything. Here's the output of `papirus-test` (with `dtoverlay=i2c-bcm2708,i2c-rtc,ds1339` set, and `sudo hwclock -r` giving the `Cannot access the Hardware Clock via any known method.` error, it seems the PiJuice RTC isn't working for me at all - I never did check when I set the PiJuice up:

```bash
pi@calipi:~ $ papirus-test
Assuming this is a PaPiRus HAT board

---- Detect temperature sensor and RTC (RTC for HAT only) ----
Other device found at: 14
NXP LM75BD detected at: 0x48
Other device found at: 68
MCP7940N detected at: 0x6f

---- Screen Information ----
Vendor: Pi Supply
Product: PaPiRus ePaper HAT v1.91
Serial: 342d3938-3462-2d38-3836-62313433327b
Date/Time from RTC: Tuesday 31 December 2019 - 15:07:28
Temperature from sensor: 31 degrees Celsius
Panel info: EPD 2.0 200 x 96 version=4 COG=2 FILM=231
```

I'm sure these two boards must be able to function together, I just don't know how to get them to play nice? Thank you for your help! :D

ChristopherRush, 2019-12-31:
Does the PiJuice work on it's own and does the eink display work on it's
own?

andrewsmyname, 2019-12-31:
The PiJuice works without issue, though the PaPiRus HAT doesn't display anything with the PiJuice disconnected. The output of `papirus-test` without the PiJuice in situ is:

```bash
pi@calipi:~ $ papirus-test
Assuming this is a PaPiRus HAT board

---- Detect temperature sensor and RTC (RTC for HAT only) ----
NXP LM75BD detected at: 0x48
MCP7940N detected at: 0x6f

---- Screen Information ----
Vendor: Pi Supply
Product: PaPiRus ePaper HAT v1.91
Serial: 342d3938-3462-2d38-3836-62313433327b
Date/Time from RTC: Tuesday 31 December 2019 - 15:38:47
Temperature from sensor: 27 degrees Celsius
Panel info: EPD 2.0 200 x 96 version=4 COG=2 FILM=231
```

ChristopherRush, 2019-12-31:
What error do you receive when running one of the PaPiRus examples? If any.

andrewsmyname, 2019-12-31:
I receive no errors, but I do note that I can't close the process with the usual CTRL+C that other cli applications respect. Perhaps it's hanging or such? Thank you for helping by the way :-)

andrewsmyname, 2019-12-31:
I did have some issues getting fuse set up, apologies I forgot to mention. When I boot it seems to be running fine now though:

'''bash
pi@calipi:~ $ sudo systemctl status epd-fuse.service
● epd-fuse.service - A Fuse driver for the Repaper.org EPD
   Loaded: loaded (/usr/lib/systemd/system/epd-fuse.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2019-12-31 15:39:52 GMT; 19min ago
  Process: 598 ExecStartPre=/usr/bin/env mkdir -p ${EPD_MOUNTPOINT} (code=exited, status=0/SUCCESS)
  Process: 643 ExecStart=/usr/sbin/epd_fuse --panel=${EPD_SIZE} $EPD_OPTS ${EPD_MOUNTPOINT} (code=exited, status=0/SUCCESS)
 Main PID: 675 (epd_fuse)
    Tasks: 4 (limit: 4915)
   Memory: 852.0K
   CGroup: /system.slice/epd-fuse.service
           └─675 /usr/sbin/epd_fuse --panel=2.0 -o allow_other -o default_permissions /dev/epd

Dec 31 15:39:51 calipi systemd[1]: Starting A Fuse driver for the Repaper.org EPD...
Dec 31 15:39:52 calipi systemd[1]: Started A Fuse driver for the Repaper.org EPD.
'''

Though if I stop the service and attempt to restart it, I get an error:
'''bash
pi@calipi:~ $ sudo systemctl status epd-fuse.service
● epd-fuse.service - A Fuse driver for the Repaper.org EPD
   Loaded: loaded (/usr/lib/systemd/system/epd-fuse.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2019-12-31 15:59:37 GMT; 4s ago
  Process: 2403 ExecStartPre=/usr/bin/env mkdir -p ${EPD_MOUNTPOINT} (code=exited, status=1/FAILURE)

Dec 31 15:59:37 calipi systemd[1]: Starting A Fuse driver for the Repaper.org EPD...
Dec 31 15:59:37 calipi env[2403]: mkdir: cannot create directory ‘/dev/epd’: File exists
Dec 31 15:59:37 calipi systemd[1]: epd-fuse.service: Control process exited, code=exited, status=1/FAILURE
Dec 31 15:59:37 calipi systemd[1]: epd-fuse.service: Failed with result 'exit-code'.
Dec 31 15:59:37 calipi systemd[1]: Failed to start A Fuse driver for the Repaper.org EPD.
'''

Creating the directory manually with `sudo fusermount -u /dev/epd` solves this and I can restart the service. Very odd, though...

Hope some of this info helps?

tvoverbeek, 2019-12-31:
@andrewsmyname It seems there have been changes to libfuse in the Debian buster which cause the papirus programs to hang. Will look into this, but no guarantees on a time-line (First new PiJuice software release ...)

ryanteck, 2019-12-31:
@shawaj I don't currently have a Pijuice, if you send one over I can always look at the software to fix it?

andrewsmyname, 2020-01-01:
> @andrewsmyname It seems there have been changes to libfuse in the Debian buster which cause the papirus programs to hang. Will look into this, but no guarantees on a time-line (First new PiJuice software release ...)

Ah I see, does this mean that the PaPiRus product is essentially unusable with the current Pi4 raspbian, then? :-(

tvoverbeek, 2020-01-05:
@andrewsmyname @ryanteck @ChristopherRush 
Found he problem with the Pi4. Since it uses a differrent SOC which is 64-bit the peripheral base address has changed to 0xfe000000 (was 0x3f000000 for Pi2 etc.). WIll try to modify the driver code to get this address automatically, so the code will run on all Pi models.

tvoverbeek, 2020-01-06:
@andrewsmyname Driver software is updated on the repaper/gratis repository. Now it works for Pi4.
Suggest to reinstall the PaPiRus software, but first to remove the existing software:
```
sudo systemctl stop epd-fuse
sudo systemctl disable epd-fuse
sudo rm /usr/sbin/epd_fuse
sudo rm /usr/local/bin/papirus-*
sudo rm -fr /usr/local/lib/python*/dist-packages/papirus*
```
then run the installation again:
```
curl -sSL https://pisupp.ly/papiruscode | sudo bash
```

andrewsmyname, 2020-01-10:
Thank you very much @tvoverbeek that fixed it! :)

closed 2020-01-10 by andrewsmyname