PiSupply/PaPiRus #215: Has this been tested on the latest 5.4 kernels?

pelwell, 2020-10-02:
Hi,

We've noticed that a number of displays have stopped working with the switch to the 5.4 kernel. This is caused by a change to the use of GPIO descriptors by some of the display drivers (mainly fbtft, but their might be others). Have you tested the PaPiRus displays on rpi-5.4.y? If you haven't and you find that it doesn't work, try changing the `reset-gpios` property in the overlay to `reset-gpios = <&gpio 24 1>;`.

shawaj, 2020-10-02:
@ChristopherRush or @ryanteck could you give this a try?

ryanteck, 2020-10-02:
Will do shortly,

IRC we don't use fbtft so it shouldn't be affected.

ChristopherRush, 2020-10-02:
Will boot one up now

ryanteck, 2020-10-02:
@ChristopherRush Ok then I'll leave it.

shawaj, 2020-10-02:
The only thing that uses framebuffer is the fbcopy script https://github.com/PiSupply/PaPiRus/blob/master/bin/papirus-fbcopy

Other than that it's the EPD fuse stuff in rePaper/gratis repo

pelwell, 2020-10-02:
I took a brief look at the repaper driver after posting and I _think_ it's OK - unlike fbtft, the knowledge that reset is active-low seems to be built into the code, not learnt from Device Tree.

shawaj, 2020-10-02:
@pelwell thanks very much for the heads up!

shawaj, 2020-10-02:
@ChristopherRush if you have time can you test the media center hat as well?

ChristopherRush, 2020-10-02:
@shawaj Yes will do

ChristopherRush, 2020-10-02:
@pelwell How did you test the display?

pelwell, 2020-10-02:
I didn't - I asked the question because a number of other displays have had problems under 5.4.

I've already speculatively fixed the Media Centre overlay - see https://github.com/raspberrypi/linux/commit/b7d685c0b1bd1b98af0e9c1f5d43769982bdbfb2#diff-dc041e47200262dfab1ff9f9e0ef64c7

ChristopherRush, 2020-10-02:
@shawaj @pelwell @ryanteck So far it is working fine. I'm using a Pi 4 with the PaPiRus HAT and 2.7" display. Kernel version is 5.4.51 and I have tested the following:
papirus-clear
papirus-clock
papirus-fbcopy
papirus-snake

pelwell, 2020-10-02:
If it was affected then it would have been dead-on-arrival - this isn't a subtle bug.

shawaj, 2020-10-02:
@pelwell guessing for hard-coded eeprom based dt blobs they'll need to override etc the new overlay in config.txt ?

Will make sure new batch uses new overlay

pelwell, 2020-10-02:
Unfortunately yes - if the EEPROM includes the overlay content then you are going to need to apply the overlay. I now recommend putting the name of the overlay in the EEPROM rather than the .dtbo itself - the firmware will load the current version.

shawaj, 2020-10-02:
@pelwell ah perfect, so does that just go in the settings.txt? Or elsewhere?

pelwell, 2020-10-02:
That would have been nice, wouldn't it, but no. Instead, you put the name of the overlay - in this case "papirus" - in a .txt file and use that in place of `papirus.dtbo`:
```
$ echo papirus > myoverlay_name.txt
$ eepmake mysettings.txt myhat.eep myoverlay_name.txt
```
Any trailing whitespace is stripped, so don't worry about newlines etc.

This is covered in the current version of https://github.com/raspberrypi/hats/blob/master/devicetree-guide.md.

shawaj, 2020-10-02:
@pelwell does the name of the .txt file have to be the same as the overlay too? Or just the contents? 

i.e. should is be called papirus.txt as well or can it be anything?

pelwell, 2020-10-02:
Anything you like - it's just a source of the bytes that get merged.

shawaj, 2020-10-02:
@pelwell awesome, thanks very much that's very useful to know. Have a great weekend :-)

tvoverbeek, 2020-10-19:
@pelwell Phil, Your  changes to the media-center-overlay.dts work.
Without them (using the original overlay) I get a black screen without backlight on.
When can we expect them in apt update?

pelwell, 2020-10-19:
I don't have a date at the moment, but the SPI CS changes give us a reason to push one or two more 5.4 kernels into apt before switching to 5.9/5.10.

pelwell, 2020-10-22:
The latest rpi-update firmware has also been pushed to the "stable" branch (`sudo BRANCH=stable rpi-update`). If no problems are reported overnight it we'll probably bump the apt packages tomorrow.

tvoverbeek, 2020-10-22:
Thanks Phil. WIll keep an eye on apt update

pelwell, 2020-10-23:
The update should be live round-about now.

tvoverbeek, 2020-10-24:
Thanks. Latest apt update (22 Oct) has the corrcted media-overlay

closed 2020-10-24 by tvoverbeek