PiSupply/PaPiRus #216: Open source hardware certification

shawaj, 2020-10-02:
PaPiRus HAT certification
[image]
[OSHW_mark_UK000017.pdf](https://github.com/PiSupply/PaPiRus/files/5320738/OSHW_mark_UK000017.pdf)

shawaj, 2020-10-02:
PaPiRus Zero certification
[image]
[OSHW_mark_UK000018.pdf](https://github.com/PiSupply/PaPiRus/files/5320741/OSHW_mark_UK000018.pdf)

closed 2020-10-02 by shawaj

ChristopherRush, 2020-10-05:
[image]

ChristopherRush, 2020-10-05:
[image]