PiSupply/PaPiRus #217: Invert Display Colors?

ahnyerkeester, 2021-04-27:
I'm using Python and would like to have a black background with white letters. Is there a way to do that? I've been looking through examples and such and can't find it.  Thanks in advance.

tvoverbeek, 2021-04-28:
Prepare the image as a negative image (white on black) with PIL (Python Image Library)

ahnyerkeester, 2021-04-28:
I'm displaying text, not images. Will that still work?

tvoverbeek, 2021-04-28:
See the papirus-write example (https://github.com/PiSupply/PaPiRus/tree/master/bin/papirus-write) which calls the PapirusTextPos (https://github.com/PiSupply/PaPiRus/blob/master/papirus/textpos.py) for how to do this.

ahnyerkeester, 2021-04-28:
Thanks @tvoverbeek, I'm fishing through textpos.py right now! Thank you for affirming that I'm barking up the correct tree.