PiSupply/PaPiRus #219: Installing epd on Raspberry Pi (Ubuntu 20.04 server)

Jct9000, 2021-05-04:
I'm trying to install epd on a Raspberry Pi running Ubuntu 20.04 server. I this make command and coming up with this error. 
Please help.

sudo make rpi EPD_IO=epd_io.h PANEL_VERSION='V231_G2'
make DESTDIR="" PREFIX=/usr SERVICE=systemd PLATFORM=../RaspberryPi PANEL_VERSION="V231_G2" EPD_IO="epd_io.h" -C PlatformWithOS/driver-common
make[1]: Entering directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"' -lfuse -pthread -lrt -L/opt/vc/lib -lbcm_host -o "gpio_test" gpio_test.o gpio.o
/usr/bin/ld: gpio.o: in function `GPIO_setup':
gpio.c:(.text+0xc): undefined reference to `bcm_host_get_peripheral_address'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:127: gpio_test] Error 1
make[1]: Leaving directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
make: *** [Makefile:84: rpi] Error 2

tvoverbeek, 2021-05-04:
It seems it cannot find libbcm_host.so which contains the bcm_host_get_peripheral_address function.
On Raspberry Pi OS this library is in /opt/vc/lib.
You have to find out where it is located on Ubuntu 20.04 server and change the `-L/opt/vc/lib -lbcm_host` in the Makefile accordingly.

Jct9000, 2021-05-04:
I found the lib in /usr/lib/aarch64-linux-gnu and changed it in the Makefile. I'm still getting an error. The Ubuntu 20.04 version is 64 bit. Is that causing the error? Do I have to reinstall the 32 bit OS version?

sudo make rpi EPD_IO=epd_io.h PANEL_VERSION='V231_G2'
make DESTDIR="" PREFIX=/usr SERVICE=systemd PLATFORM=../RaspberryPi PANEL_VERSION="V231_G2" EPD_IO="epd_io.h" -C PlatformWithOS/driver-common
make[1]: Entering directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"' -lfuse -pthread -lrt -L/usr/lib/aarch64-linux-gnu -lbcm_host -o "gpio_test" gpio_test.o gpio.o
/usr/bin/ld: gpio.o: in function `GPIO_setup':
gpio.c:(.text+0xc): undefined reference to `bcm_host_get_peripheral_address'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:127: gpio_test] Error 1
make[1]: Leaving directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
make: *** [Makefile:84: rpi] Error 2

tvoverbeek, 2021-05-04:
It still does not find the function.
Can you check if it is in the library?
`nm /usr/lib/aarch6464-linux-gnu/libbcmhost.so | grep bcm_host_get_peripheral_address`
Does /opt/vc/include exist?

Jct9000, 2021-05-04:
nm /usr/lib/aarch6464-linux-gnu/libbcmhost.so | grep bcm_host_get_peripheral_address
nm: '/usr/lib/aarch6464-linux-gnu/libbcmhost.so': No such file

... and /opt/lib/include does exist. It is linked to /usr/include (vc -> /usr).

tvoverbeek, 2021-05-05:
I am guessing you are using either a Pi 3B(+) or Pi 4B (since you are on a 64-bit OS).
Since bcm_host_get_peripheral_address() seems not to be available in Ubuntu 20.04, change line 246 in gpio.c
from `base_address = bcm_host_get_peripheral_address();` to:
`base_address = 0x3F000000` for Pi 3B(+)
`base_address = 0xFE000000` for Pi 4B

See also https://www.raspberrypi.org/documentation/hardware/raspberrypi/peripheral_addresses.md

Jct9000, 2021-08-31:
I had to put this project down for a few months but I'm back to see if it can be resurrected.  I have reinstalled Ubuntu 20.04 server (32 bit .. this time) and tried to install the Papirus software. I followed your instructions above and got a little further .... I think. 

This is error message I'm getting after updating the base_address to 0x3F000000 and running the make command: 

make rpi EPD_IO=epd_io.h PANEL_VERSION='V231_G2'
make DESTDIR="" PREFIX=/usr SERVICE=systemd PLATFORM=../RaspberryPi PANEL_VERSION="V231_G2" EPD_IO="epd_io.h" -C PlatformWithOS/driver-common
make[1]: Entering directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"'   -c -o gpio_test.o gpio_test.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"'   -c -o gpio.o ../RaspberryPi/gpio.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"' -lfuse -pthread -lrt -L/opt/vc/lib -lbcm_host -o "gpio_test" gpio_test.o gpio.o
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"'   -c -o epd_test.o epd_test.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"'   -c -o spi.o spi.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"'   -c -o epd.o V231_G2/epd.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"' -lfuse -pthread -lrt -L/opt/vc/lib -lbcm_host -o "epd_test" epd_test.o gpio.o spi.o epd.o
/usr/bin/ld: epd.o: in function `EPD_create':
epd.c:(.text+0x32): undefined reference to `timer_create'
/usr/bin/ld: epd.o: in function `frame_fixed_repeat':
epd.c:(.text+0xe20): undefined reference to `timer_settime'
/usr/bin/ld: epd.c:(.text+0xe50): undefined reference to `timer_gettime'
/usr/bin/ld: epd.o: in function `frame_data_repeat':
epd.c:(.text+0xf24): undefined reference to `timer_settime'
/usr/bin/ld: epd.c:(.text+0xf54): undefined reference to `timer_gettime'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:132: epd_test] Error 1
make[1]: Leaving directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
make: *** [Makefile:84: rpi] Error 2

tvoverbeek, 2021-08-31:
You have undefined references to timer_create, timer_settime and timer_gettime.
These are POSIX timer functions. Normally they should be enabled, but Ubuntu can configure the kernel without support for POSIX timers.
See http://manpages.ubuntu.com/manpages/focal/en/man2/timer_create.2.html for some details.
In short: find out where these functions are located.