PiSupply/PaPiRus #220: Doesn't work on Pi Zero W

frioux, 2021-08-17:
I think this is a build / installation error, but it's hard to tell.  I started off using `curl -sSL https://pisupp.ly/papiruscode | sudo bash`, which seemed to complete successfully, as far as I could tell, but then when I tried to actually do anything I'd get errors:

```
$ sudo /usr/local/bin/papirus-set
Failed to stop epd-fuse.service: Unit epd-fuse.service not loaded.
/usr/local/bin/papirus-set: line 20: /etc/default/epd-fuse: No such file or directory
sed: can't read /etc/default/epd-fuse: No such file or directory

sed: can't read /etc/default/epd-fuse: No such file or directory
Failed to start epd-fuse.service: Unit epd-fuse.service not found.
```

I figured this was something with the driver so I started following the instructions to build that and ended up here:

```
$ make rpi EPD_IO=epd_io.h PANEL_VERSION='V231_G2'
make DESTDIR="" PREFIX=/usr SERVICE=systemd PLATFORM=../RaspberryPi PANEL_VERSION="V231_G2" EPD_IO="epd_io.h" -C PlatformWithOS/driver-common
make[1]: Entering directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"' -lfuse -pthread -lrt -L/opt/vc/lib -lbcm_host -o "gpio_test" gpio_test.o gpio.o
/usr/bin/ld: gpio.o: in function `GPIO_setup':
gpio.c:(.text+0x14): undefined reference to `bcm_host_get_peripheral_address'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:127: gpio_test] Error 1
make[1]: Leaving directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
make: *** [Makefile:84: rpi] Error 2
```

Let me know how I can help

tvoverbeek, 2021-08-17:
The same make you are running is also executed by the install script, so something is wrong somewhere.
In your make the linker cannot find the bcm_host_get_peripheral_address function. this function is located in the
bcm_host shared library in the /opt/vc/lib directory.
This is specified in your make command line.
So I suspect the bcm_host library is missing.
What is the output from `ls -l /opt/vc/lib`? Does bcm_host.so show up there?

frioux, 2021-08-17:
looks like it's there to me?

frioux, 2021-08-17:
[image]
  woops forgot to attach screenshot

frioux, 2021-08-17:
Oh wait so it's renamed from `bcm_host.so` to `libbcm_host.so`, I guess?

tvoverbeek, 2021-08-17:
No, link spec like `-lbcm_host` means look for libbcm_host.a or libbcm_host.so

frioux, 2021-08-17:
A friend pointed out that this works if I change this line:

```
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"' -lfuse -pthread -lrt -L/opt/vc/lib -lbcm_host -o "gpio_test" gpio_test.o gpio.o
```

to this (move `-lbcm_host` to the end)

```
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"' -lfuse -pthread -lrt -L/opt/vc/lib -o "gpio_test" gpio_test.o gpio.o -lbcm_host
```

I'm not sure how to submit that as a PR or I would

tvoverbeek, 2021-08-18:
That is not needed. Should work as is.
Remove recursilvelu /tmp/papirus/gratis and run the `git clone https://github.com/repaper/gratis.git` in /tmp/papirus again.
Then cd to /tmp/papirus/gratis and tun the make again.
Here is what I got on a PiZero:
```
pi@adafruitio:/tmp/papirus/gratis $ make rpi EPD_IO=epd_io.h PANEL_VERSION='V231_G2'
make DESTDIR="" PREFIX=/usr SERVICE=systemd PLATFORM=../RaspberryPi PANEL_VERSION="V231_G2" EPD_IO="epd_io.h" -C PlatformWithOS/driver-common
make[1]: Entering directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"'   -c -o gpio_test.o gpio_test.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"'   -c -o gpio.o ../RaspberryPi/gpio.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"' -lfuse -pthread -lrt -L/opt/vc/lib -lbcm_host -o "gpio_test" gpio_test.o gpio.o
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"'   -c -o epd_test.o epd_test.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"'   -c -o spi.o spi.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"'   -c -o epd.o V231_G2/epd.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"' -lfuse -pthread -lrt -L/opt/vc/lib -lbcm_host -o "epd_test" epd_test.o gpio.o spi.o epd.o
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"'   -c -o epd_fuse.o epd_fuse.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"' -lfuse -pthread -lrt -L/opt/vc/lib -lbcm_host -o "epd_fuse" epd_fuse.o gpio.o spi.o epd.o
make[1]: Leaving directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
pi@adafruitio:/tmp/papirus/gratis $ 
```
No errors.

Alternatively run papirus-setup again.

frioux, 2021-08-18:
```
pi@pizero:~ $ cd /tmp
pi@pizero:/tmp $ uname -a
Linux pizero 5.10.52+ #1441 Tue Aug 3 18:08:18 BST 2021 armv6l GNU/Linux
pi@pizero:/tmp $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye
pi@pizero:/tmp $ git clone https://github.com/repaper/gratis.git
Cloning into 'gratis'...
remote: Enumerating objects: 2620, done.
remote: Counting objects: 100% (8/8), done.
remote: Compressing objects: 100% (8/8), done.
remote: Total 2620 (delta 0), reused 1 (delta 0), pack-reused 2612
Receiving objects: 100% (2620/2620), 8.35 MiB | 960.00 KiB/s, done.
Resolving deltas: 100% (1365/1365), done.
pi@pizero:/tmp $ cd gratis/
pi@pizero:/tmp/gratis $ make rpi EPD_IO=epd_io.h PANEL_VERSION='V231_G2'
make DESTDIR="" PREFIX=/usr SERVICE=systemd PLATFORM=../RaspberryPi PANEL_VERSION="V231_G2" EPD_IO="epd_io.h" -C PlatformWithOS/driver-common
make[1]: Entering directory '/tmp/gratis/PlatformWithOS/driver-common'
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"'   -c -o gpio_test.o gpio_test.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"'   -c -o gpio.o ../RaspberryPi/gpio.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"' -lfuse -pthread -lrt -L/opt/vc/lib -lbcm_host -o "gpio_test" gpio_test.o gpio.o
/usr/bin/ld: gpio.o: in function `GPIO_setup':
gpio.c:(.text+0x14): undefined reference to `bcm_host_get_peripheral_address'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:127: gpio_test] Error 1
make[1]: Leaving directory '/tmp/gratis/PlatformWithOS/driver-common'
make: *** [Makefile:84: rpi] Error 2
```

tvoverbeek, 2021-08-18:
Thanks for including the lsb_release info:
```
pi@pizero:/tmp $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 11 (bullseye) <---
Release:        11  <---
Codename:       bullseye
```
There is no officially supported Raspberry Pi OS based on Debian 11 (bullseye) yet,
Raspberry Pi OS is still on Debian 10 (buster).
So you are  basically on your own if you need to run under bullseye.
The bcm_host library is Pi specific so probably not part of Debian 11.

You could replace the call to bcm_host_get_peripheral_address with the hard coded value for the Pi Zero.
See the official  documentation for the actual value.

frioux, 2021-08-18:
Well, FWIW when I put the `-lbcm_host` (and other `-l` flags) at the end I was able to build everything, get it installed, and actually use the device, so I still think that the suggestion I made before will need to be implemented once bullseye is officially released.

Utopiah, 2021-12-20:
FWIW if that's a proper fix then it could mean moving `${LDFLAGS}` at the end of the compilation line in https://github.com/repaper/gratis/blob/master/PlatformWithOS/driver-common/Makefile#L118

tvoverbeek, 2021-12-22:
The fix is now in the repaper/gratis repository.
Turns out in Bullseye the bcm_host.h and libbcm_host.a files are not in the /opt/vc directory tree.
The whole tree has been removed (no more proprietary video core support in Bullseye).
The files are now in the standard include and library directories.
Moving ${LDFLAGS} to the end makes it work on both Buster and Bullseye.
Closing the issue (and @Utopiah for sorting this out).

closed 2023-02-18 by frioux