PiSupply/PaPiRus #222: Current Raspbian support (bullseye, released October 30th 2021)

Utopiah, 2021-12-20:
Hi, Auto Installation doesn't work on the current RPi OS. Manual compilation also fails.

Few months ago this was identified https://github.com/PiSupply/PaPiRus/issues/220#issuecomment-901421426 but since nearly 2 months now [the official RPi distribution points to something not compatible anymore](https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-32-bit).

Utopiah, 2021-12-20:
1. moving ${LDFLAGS} at the end of the compilation line in https://github.com/repaper/gratis/blob/master/PlatformWithOS/driver-common/Makefile#L118 
2. manually installing the driver as described in https://github.com/PiSupply/PaPiRus#install-driver-option-2-best-to-run-all-of-these-commands-as-root-using-sudo but relying on that modified Makefile
3. and finally installing the API for Python3 according to https://github.com/PiSupply/PaPiRus#install-python-api-best-to-run-all-of-these-commands-as-root-using-sudo

made the Python3 example work on Raspbian 11 (bullseye) on a RPi3.

tvoverbeek, 2021-12-22:
Fix is now in the repaper/gratis repository, so standard install should work now.
See also https://github.com/PiSupply/PaPiRus/issues/220#issuecomment-999855458
Closing the issue.

closed 2021-12-22 by tvoverbeek