PiSupply/PaPiRus #223: Unit file epd-fuse.service does not exist.

ericbrian, 2021-12-21:
I ran the script as the readme says:

    curl -sSL https://pisupp.ly/papiruscode | sudo bash

And then even ran the script manually with `sudo papirus-setup` from ``/usr/local/bin`

This second manual run resulted in 

```
pi@RPi-PiHole:/usr/local/bin $ sudo papirus-setup
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libfuse-dev is already the newest version (2.9.9-5).
0 upgraded, 0 newly installed, 0 to remove and 26 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
fonts-freefont-ttf is already the newest version (20120503-10).
0 upgraded, 0 newly installed, 0 to remove and 26 not upgraded.
rm: cannot remove '/tmp/papirus': No such file or directory
Cloning into 'gratis'...
remote: Enumerating objects: 2620, done.
remote: Counting objects: 100% (8/8), done.
remote: Compressing objects: 100% (8/8), done.
remote: Total 2620 (delta 0), reused 1 (delta 0), pack-reused 2612
Receiving objects: 100% (2620/2620), 8.35 MiB | 1.02 MiB/s, done.
Resolving deltas: 100% (1365/1365), done.
make DESTDIR="" PREFIX=/usr SERVICE=systemd PLATFORM=../RaspberryPi PANEL_VERSION="V231_G2" EPD_IO="epd_io.h" -C PlatformWithOS/driver-common
make[1]: Entering directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"'   -c -o gpio_test.o gpio_test.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"'   -c -o gpio.o ../RaspberryPi/gpio.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I/opt/vc/include -I. -DEPD_IO='"epd_io.h"' -lfuse -pthread -lrt -L/opt/vc/lib -lbcm_host -o "gpio_test" gpio_test.o gpio.o
/usr/bin/ld: gpio.o: in function `GPIO_setup':
gpio.c:(.text+0x14): undefined reference to `bcm_host_get_peripheral_address'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:127: gpio_test] Error 1
make[1]: Leaving directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
make: *** [Makefile:84: rpi] Error 2
make DESTDIR="" PREFIX=/usr SERVICE=systemd PLATFORM=../RaspberryPi PANEL_VERSION="V231_G2" EPD_IO="epd_io.h" -C PlatformWithOS/driver-common install
make[1]: Entering directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
[ -e epd_fuse ] || false
make[1]: *** [Makefile:55: check-compiled] Error 1
make[1]: Leaving directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
make: *** [Makefile:87: rpi-install] Error 2
Failed to enable unit: Unit file epd-fuse.service does not exist.
Failed to stop epd-fuse.service: Unit epd-fuse.service not loaded.
/usr/local/bin/papirus-set: line 20: /etc/default/epd-fuse: No such file or directory
sed: can't read /etc/default/epd-fuse: No such file or directory

2.0
sed: can't read /etc/default/epd-fuse: No such file or directory
Failed to start epd-fuse.service: Unit epd-fuse.service not found.
pi@RPi-PiHole:/usr/local/bin $
```

tvoverbeek, 2021-12-21:
See https://github.com/PiSupply/PaPiRus/issues/222#issuecomment-998249582 for now to get itworking on Bullseye.

ericbrian, 2021-12-21:
Sadly, it did not work. I got the exact same result.

Utopiah, 2021-12-21:
@ericbrian if you re-run `sudo papirus-setup` it's going to overwrite the manual fixed `papirus` build so make you sure you do the steps manually for now, not via the setup.