PiSupply/PaPiRus #224: Error Trying To Run First Test

nateProjects, 2022-05-10:
Hiya, This seemed to install fine, but I've had no luck getting it to run:
```
$ papirus-test
Traceback (most recent call last):
  File "/usr/local/bin/papirus-test", line 69, in <module>
    papirus = Papirus()
  File "/usr/local/lib/python3.9/dist-packages/papirus/epd.py", line 65, in __init__
    self._lm75b = LM75B()
  File "/usr/local/lib/python3.9/dist-packages/papirus/lm75b.py", line 24, in __init__
    self._bus.write_byte_data(self._address, LM75B_CONF_REGISTER, LM75B_CONF_NORMAL)
OSError: [Errno 121] Remote I/O error

```
Raspberry Pi Zero W 2 / Raspbian 11 (bullseye) / PapiRus Zero v1.2 2.0"

nateProjects, 2022-05-13:
It turned out to be a fault Pi Zero - swapped it out with another one and it worked! 

Thanks

tvoverbeek, 2022-05-13:
OK. Closing the issue

closed 2022-05-13 by tvoverbeek