PiSupply/PaPiRus #229: Screen status: Unsupported COG

orypersec, 2023-03-17:
When i run papirus-system command i get "Unsupported Cog" error. I have raspberry pi zero w with pwnagotchi installed.

tvoverbeek, 2023-03-17:
COG = Chip On Glass. It seems not to detect the expected EPD display.
How did you install the software drivers for the PaPiRus?
Pwnagorchi has its own code for PaPiRus, but you still need the device drivers.
Do /dev/epd and /dev/spi0.0 exist?

tvoverbeek, 2023-03-17:
That is my bad. /dev/spidev0.0 is correct.
What is the contents of /etc/default/epd-fuse?

orypersec, 2023-03-17:
```
# Default settings for epd-fuse file is sourced by /bin/sh from
# /etc/init.d/epd-fuse

# Options to pass to epd_fuse
#EPD_MOUNTPOINT=/dev/epd
EPD_SIZE=2.0
#EPD_OPTS='-o allow_other -o default_permissions'
```

tvoverbeek, 2023-03-17:
That is as expected. The display is the 200x96 one (2.0 inch) which fits the pi zero w?
If so, check the papirus pHat is correctly conencted to the 40-pin header. It is easy to be off by 1 pin or 1 row.

orypersec, 2023-03-17:
Yes it is 2.0 inch. I soldered the pins today. I connected all 40 pins. If it's because of the pins I can solder it again.

tvoverbeek, 2023-03-17:
That is the only thing I can think off right now. Please double check the solder connections.

orypersec, 2023-03-17:
Thank you very much, I will try it tomorrow and share the result.

orypersec, 2023-03-17:
I installed raspberry OS and the screen did not work again. Now I reinstalled pwnagotchi OS and the screen worked. I don't understand how it worked but it did. Thanks for everything.

closed 2023-03-17 by orypersec